dotnet-bluetooth-le/dotnet-bluetooth-le #107: Multiple notifcation for devices with same characteristic id's

xabre, 2016-11-15:
While thinking about the implementation of #90 I noticed the following case:

- Suppose we are connected to two devices of the same make at the same time (same services and characteristic ids) . 
- We start notifcations for **device1**(d1) -> s -> characteristic(c) and subscribe with **handler1** (h1)
- We start notifcations for **device2**(d2) -> s -> characteristic(c) and subscribe with **handler2** (h2)
- Basically we subscribe for notifcations on the same characteristic on both devices.
- We receive a notification only from **d1** -> c and both are handlers (h1 and h2) are invoked because we only filter by characteristic id an not device id when raising value changed events.

@smstuebe what do you think?

smstuebe, 2016-11-15:
let us have a look at this in office :) Currently preparing some talks, xamarin dev days and some other community stuff ^^

xabre, 2017-04-24:
#144 should solve this

closed 2017-04-25 by xabre