dotnet-bluetooth-le/dotnet-bluetooth-le #112: Bluetooth state of iOS device is 'Unknown' for first time access

leelatoo, 2016-11-17:
Hi,

We access the bluetooth state of iOS device using `CrossBluetoothLE.Current.State`.

Bluetooth is turned on in iOS device. When we access the bluetooth state using the above mentioned API, it returns 'Unknown' for the first time.(but iOS device's bluetooth is turned on).

When we access the same API again, it returns proper state 'On'.

Is this the proper way of accessing the bluetooth state of the device.?
or, are we missing anything here?

Can you please help us?

smstuebe, 2016-11-17:
hi,

We provided a issue template... 
https://github.com/xabre/xamarin-bluetooth-le/blob/master/ISSUE_TEMPLATE.md
Pls provide the Information like device, version, os, ... 

For the next time: fill it out please.

leelatoo, 2016-11-17:
@smstuebe sure.

## Steps to reproduce

`IBluetoothLE _bluetoothLe = CrossBluetoothLE.Current;
`
1. Turn on bluetooth in iOS device. Use `_bluetoothLe.State` API to know the bluetooth status of iOS device. Register for bluetooth state change event `_bluetoothLe.StateChanged`

2. It returns the bluetooth state as 'Unknown', though bluetooth is turned on in the iOS device.

3. Now, bluetooth state changed event gets fired. the state is on.

4. Call the API `CrossBluetoothLE.Current.State` again. It returns the state as 'On' properly. 

## Expected behavior
When we access the bluetooth state for the first time, it should return the proper value as 'On'

## Actual behavior
Bluetooth state is not reflected properly for the first time.

## Configuration
1.
**Version:** 9.3.1

**Platform:** iOS

**Device:** iPhone 5S.

2.
**Version:** 10.1.1

**Platform:** iOS

**Device:** iPhone 6S.

We are using latest version of Plugin.BLE

Rob-Houweling, 2016-11-30:
I've been struggling with this as well without using a plugin.
What I think happens is that the CBCentralManager will return the Unknown state if the initialization hasn't finished yet. 
By the look of it (code of the plugin you're using) they don't init() CBCentralManager so this explains the unknown state:
protected override void InitializeNative()
{
    _centralManager = new CBCentralManager(DispatchQueue.CurrentQueue);
    _centralManager.UpdatedState += (s, e) => State = GetState();
}

AlexusBlack, 2018-04-13:
Okay, so I checked the Plugin.BLE source code and played with it a bit and found an inconsistency between Android and iOS state detection that Plugin.BLE does not fully compensate.

While on Android we can get state immediately, on iOS initial state is unknown and delivered later via state update event. This is alike CBCentralManager which also delivers states via a delegate. 

I wrote a simple workaround which is a solution for issues #188 and #112. I tested it on both iOS and Android, using iOS 11 IP7 and Android 7 Samsung Galaxy S7.

```csharp
/// <summary>
/// Checks the bluetooth adapter if it is on or off.
/// </summary>
private async Task CheckBluetooth() {
    // iniial state check
    var state = await GetBluetoothState(Plugin.BLE.CrossBluetoothLE.Current);
    if(state == BluetoothState.Off)
    {
        ShowDisabledBluetoothAlert();
    }
    
    // tracking state in case if user disable his bluetooth
    Plugin.BLE.CrossBluetoothLE.Current.StateChanged += (o, e) =>
    {
        if(e.NewState == BluetoothState.Off)
        {
            ShowDisabledBluetoothAlert();
        }
    };
}

/// <summary>
/// Get current bluetooth state
/// </summary>
/// <param name="ble">Crossplatform bluetooth implementation object</param>
/// <returns>BluetoothState</returns>
private Task<BluetoothState> GetBluetoothState(IBluetoothLE ble)
{
    var tcs = new TaskCompletionSource<BluetoothState>();

    if (ble.State != BluetoothState.Unknown)
    {
        // If we can detect state out of box just returning in
        tcs.SetResult(ble.State);
    } else
    {
        // Otherwise let's setup dynamic event handler and wait for first state update
        EventHandler<Plugin.BLE.Abstractions.EventArgs.BluetoothStateChangedArgs> handler = null;
        handler = (o, e) =>
        {
            Plugin.BLE.CrossBluetoothLE.Current.StateChanged -= handler;
            // and return it as our state
            // we can have an 'Unknown' check here, but in normal situation it should never occur
            tcs.SetResult(e.NewState);
        };
        Plugin.BLE.CrossBluetoothLE.Current.StateChanged += handler;
    }

    return tcs.Task;
}

/// <summary>
/// Show alert to user that his bluetooth is disabled
/// </summary>
private void ShowDisabledBluetoothAlert()
{
    App.ShowAlert("Bluetooth disabled", "Bluetooth required to work with BLE devices.");
}
```