dotnet-bluetooth-le/dotnet-bluetooth-le #113: Android - Device name always NULL (device discovered event)

PawelGasiorowski, 2016-11-18:
Hi,

I have a problem with the Device Name. It does not depand on device model. On all devices I have tested I get NULL device name on OnDiscovered event. On IOS there is no problem. Device names are fine. On always get NULL. Rest parameters are fine and I can connect to the device.

## Expected behavior
On both platforms OnDiscovered returns IDevice object that has name.

## Actual behavior
On IOS devices has name while on Android it is always NULL.

### Crashlog

## Configuration

**Version:** 1.1.0

**Platform:** Android

xabre, 2016-11-22:
Have you tried the sample App, does it happen with it?

xabre, 2016-11-29:
Do you have any code you can share? The sample app seems to work fine. So does our current production app.

xabre, 2016-12-23:
I'm guessing it's not an issue anymore

closed 2016-12-23 by xabre