dotnet-bluetooth-le/dotnet-bluetooth-le #114: Can not connect to ble on OnePlus 3

allbdeveloper, 2016-11-19:
Hi, 

I have a report from my customer who are using my app on OnePlus 3.
He said he can not connect to ble peripheral on OnePlus 3.

His Hawei works fine but OP3 doesn't.

I didn't received the Android version yet. 

Someone could test on OnePlus 3 if you have it?
I also should get OP3 to test on it.

Thanks.

xabre, 2016-11-19:
Had a similar problem with A Huawai P9 Lite. I installed a few other 'BLE Scanner' Apps from the play store and noticed that non of them could connect to the peripherial. The problem was not with the peripherial but with either the Huwaai BLE implementeation or the peripherial.... still not solved

allbdeveloper, 2016-11-21:
Hi, 
http://recomhub.com/blog/oneplus-3-problems-with-bluetooth-solved/

Thankfully it solved customer's problem. 
It wasn't pulgin's problem but Android bluetooth cache.

allbdeveloper, 2016-11-21:
I believe this would work on other Android phones.

closed 2016-11-21 by smstuebe