dotnet-bluetooth-le/dotnet-bluetooth-le #115: ICharacteristic.WriteAsync(byte[])   Error on Android 5.0.1

huyusong, 2016-11-19:
Caused by: android.runtime.JavaProxyThrowable: Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED.
  at Plugin.BLE.Android.Characteristic.InternalWrite (System.Byte[] data) [0x00054] in <5d0080769b114fa0ac55d0d5e8f42614>:0 
  at Plugin.BLE.Android.Characteristic+<>c__DisplayClass19_0.<WriteNativeAsync>b__0 () [0x00000] in <5d0080769b114fa0ac55d0d5e8f42614>:0 
  at Plugin.BLE.Abstractions.Utils.TaskBuilder+<FromEvent>d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext () [0x000b3] in <17233c4b6076471cb4183230850d0bba>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at Plugin.BLE.Android.Characteristic+<WriteNativeAsync>d__19.MoveNext () [0x000dc] in <5d0080769b114fa0ac55d0d5e8f42614>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at Plugin.BLE.Abstractions.CharacteristicBase+<WriteAsync>d__27.MoveNext () [0x000b1] in <17233c4b6076471cb4183230850d0bba>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at WeatherMonitor.PageTab+<SendToDevice>d__18.MoveNext () [0x00072] in <939c423dfe1e41988295c028824d3434>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<ThrowAsync>m__0 (System.Object state) [0x00000] in <a709efbb17bf4aa7b8d27ea8973c7f4c>:0 
  at Android.App.SyncContext+<Post>c__AnonStorey0.<>m__0 () [0x00000] in <b56096d9aca04f2d9c2570e7f66fd957>:0 
  at Java.Lang.Thread+RunnableImplementor.Run () [0x0000b] in <b56096d9aca04f2d9c2570e7f66fd957>:0 
  at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00009] in <b56096d9aca04f2d9c2570e7f66fd957>:0 
  at (wrapper dynamic-method) System.Object:1b37c8f1-1a72-4a48-8432-f868d2c67885 (intptr,intptr)
	at mono.java.lang.RunnableImplementor.n_run(Native Method)
	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30)
	at android.os.Handler.handleCallback(Handler.java:815)
	at android.os.Handler.dispatchMessage(Handler.java:104)
	at android.os.Looper.loop(Looper.java:194)
	at android.app.ActivityThread.main(ActivityThread.java:5572)
	... 4 more
java.lang.reflect.InvocationTargetException
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:966)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:761)

xabre, 2016-11-19:
I had a simiar problem a few weeks ago. Make sure you call WriteAsync from your **main thread**, the internal Android Write has to be called from the main thread or it will fail unfortunately.

smstuebe, 2016-11-19:
Ye we should do it. Default enabled, and configurable?

petterssonjohan, 2016-11-22:
Xabre: Could you please write an example of how this could look like?

When i connect to my peripheral i would like to write data on a charactaristic. So i have tried with this:

    Xamarin.Forms.Device.BeginInvokeOnMainThread(async () =>
                {
                    var characteristic = await GetCharacteristicById(parameters, Guid.Parse("00002A87-0000-1000-8000-00805f9b34fb"));
                    byte[] array = Encoding.UTF8.GetBytes(UserDataMock.Email);
                    characteristic.WriteAsync(array);
                });

**But i guess this should be done deeper in the business layer?** 
The above code is modified a bit from the default code you wrote and is for now hard coded for a specific service 181C(UserData)

I would like to write a e-address to this charactaristic.

**Should ReadAsync also be called from main thread?**

xabre, 2016-11-24:
Xamarin.Forms.Device.BeginInvokeOnMainThread should be enough to workaround this issue. If WriteAsync still fails it means that the main thread is not the problem... Usually when, you call Read/Write as a resoult of the user interacting with the UI it's already on the main thread....

huyusong, 2016-11-28:
Xabre:  I wrote like this :
                    Device.BeginInvokeOnMainThread(async () =>
                    {
                        var bl = await PageTab.DiscoveredService.WriteAsync(tmpSend);
                    });
but still error.......

java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:761)
Caused by: java.lang.reflect.InvocationTargetException
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:966)
	... 1 more
Caused by: android.runtime.JavaProxyThrowable: Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED.
  at Plugin.BLE.Android.Characteristic.InternalWrite (System.Byte[] data) [0x00054] in <7bece02839074605be3fd9668a0f909c>:0 
  at Plugin.BLE.Android.Characteristic+<>c__DisplayClass19_0.<WriteNativeAsync>b__0 () [0x00000] in <7bece02839074605be3fd9668a0f909c>:0 
  at Plugin.BLE.Abstractions.Utils.TaskBuilder+<FromEvent>d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext () [0x000b3] in <0fdbf1b4ae3a48c4887b46bc9db47baa>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at Plugin.BLE.Android.Characteristic+<WriteNativeAsync>d__19.MoveNext () [0x000dc] in <7bece02839074605be3fd9668a0f909c>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at Plugin.BLE.Abstractions.CharacteristicBase+<WriteAsync>d__31.MoveNext () [0x000b1] in <0fdbf1b4ae3a48c4887b46bc9db47baa>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at WeatherMonitor.PageTab+<>c__DisplayClass18_0+<<SendToDevice>b__0>d.MoveNext () [0x0006a] in <68cd3e19828d4a61aa37d1d14cc949c1>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<ThrowAsync>m__0 (System.Object state) [0x00000] in <d0f603f8185749ffb37f91a992ccbafa>:0 
  at Android.App.SyncContext+<Post>c__AnonStorey0.<>m__0 () [0x00000] in <599e0e35e02b42a88f4d7cc90ea016bb>:0 
  at Java.Lang.Thread+RunnableImplementor.Run () [0x0000b] in <599e0e35e02b42a88f4d7cc90ea016bb>:0 
  at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00009] in <599e0e35e02b42a88f4d7cc90ea016bb>:0 
  at (wrapper dynamic-method) System.Object:d09e929f-ceb3-42d0-9427-8e6aca122335 (intptr,intptr)
	at mono.java.lang.RunnableImplementor.n_run(Native Method)
	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30)
	at android.os.Handler.handleCallback(Handler.java:815)
	at android.os.Handler.dispatchMessage(Handler.java:104)
	at android.os.Looper.loop(Looper.java:194)
	at android.app.ActivityThread.main(ActivityThread.java:5572)
	... 4 more
java.lang.reflect.InvocationTargetException
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:966)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:761)
[... cut ...]
	at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30)
	at android.os.Handler.handleCallback(Handler.java:815)
	at android.os.Handler.dispatchMessage(Handler.java:104)
	at android.os.Looper.loop(Looper.java:194)
	at android.app.ActivityThread.main(ActivityThread.java:5572)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:966)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:761)

xabre, 2016-11-28:
We need more information. Device type? Android version? etc...
Did you try the sample ble explorer app to connect/write?
Did you try a third party ble explorer app from the playstore to test connection and read/write? (Try 'Ble Scanner')

mwerickson, 2016-12-29:
I am receiving the same error.  I changed the code in the PCL project to run on the main thread and still have same issue.  PCL and Android project are using the same versions of Plugin.BLE.  The code IS working on iOS.  In addition to the write failed, I'm also getting descriptor write failed.  In each case, the .CanWrite was true.  Are there any workarounds or solutions for this?

Plugin.BLE: 1.2.0-beta4
Device: Nexus 5
OS: Android 6.0.1
IDE: Visual Studio 2015 R3
Xamarin: 4.2.2.6
Xamarin.Android: 7.0.2.42

Jelmer7, 2017-02-03:
Any progress on this?, i'm getting similar problems with an rfduino unit.

The unit controls an LED panel wich changes on button pushes. 
The unit has to disconnect bluetooth after each characteristic write, so i build the connection on the button clicked. First write goes fine, second write gives
`Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED.` 
while characteristic.CanWrite is true.

`private void btn_clicked(Plugin.BLE.Abstractions.Contracts.IDevice device)
        {
            Xamarin.Forms.Device.BeginInvokeOnMainThread(async () =>
            {
                adapter = CrossBluetoothLE.Current.Adapter;
                try
                {
                    await adapter.ConnectToDeviceAsync(device);
                }
                catch (DeviceConnectionException e)
                {
                    Debug.WriteLine("Connection failed");
                }
                
                var service = await device.GetServiceAsync(Guid.Parse("00002220-0000-1000-8000-00805f9b34fb"));
                var characteristic = await service.GetCharacteristicAsync(Guid.Parse("00002222-0000-1000-8000-00805f9b34fb"));

                string input = "4";
                byte[] array = Encoding.UTF8.GetBytes(input);
                var send = await characteristic.WriteAsync(array);
            });
        }`

Plugin version: 1.1.0
Device: Samsung Galaxy s6 Edge
OS: Android 6.0.1
Xamarin: 4.2.2.11
Xamarin.Android: 7.0.2.42