dotnet-bluetooth-le/dotnet-bluetooth-le #116: StartScanningForDevicesAsync with Service GUID issue

alberk8, 2016-11-21:
The StartScanningForDevicesAsync with Service GUID will never return any result in Android 4.4.  The bluetooth device is OK as it is tested with Device running Lollipop and Marshmallow.  Scanning without the Service GUID will result in all discoverable bluetooth devices in Android.  Could it be due the the lack of API from the KitKat BLE implementation as indicated in the following site?

https://github.com/iDevicesInc/SweetBlue/wiki/Android-BLE-Issues  

## Steps to reproduce

1.  (plain vanilla Android implementation)

```
 try
            {
                edText.Text = "Start Scanning ";
                await _bleAdapter.StartScanningForDevicesAsync(new[] { Guid.Parse("00000af0-0000-1000-8000-00805f9b34fb") });
              
            }
            catch (DeviceConnectionException ex)
            {
                Toast.MakeText(this, "Device Connection Exception :"  + ex.Message, ToastLength.Short).Show();
            }

```

2.

3.

## Expected behavior
Find the Device with the Service GUID

## Actual behavior
on Android 4.4 it will not be able to locate the device.  Works Ok on lollipop and Marshmallow

### Crashlog
not crash occur

## Configuration

**Version:** 1.1.0

**Platform:** Android..

**Device:**  Lenovo A6000

xabre, 2016-11-23:
This might be a Android 4.3/4.4 issue ...
As a workaround you can always filter devices by advertisement data...

alberk8, 2016-11-27:
Thanks.  End up filtering the Device.Name for KitKat.  While not ideal but works for now.

xabre, 2016-11-28:
@alberk8 you can filter by advertisement data which contains the service guid. Might be better than name.

lucafabbri, 2018-02-10:
This happens on iOS as well

erdmenchen, 2018-03-02:
Can verify the issue for android 8.1...

majdaneq, 2018-11-07:
Same issue for android 8.0