dotnet-bluetooth-le/dotnet-bluetooth-le #117: Can't get device's name on iOS when screen is locked.

KolesnikValentine, 2016-11-23:
In iOS version of my application when DeviceDiscovered event is triggered and I have my screen locked I can't get broadcasted device name through e.Device.Name because it returns me the name of my PC's bluetooth, even though it's turned off. After unlock of an iPhone it gets broadcasted names like it should. On Android everything works like a charm!

## Steps to reproduce

1. Start scanning async for BLE devices with specific UUID's.

2. Lock the screen of iPhone.

3. Log names of BLE device's names which had been found.

4. Unlock the iPhone.

5. Log device's names once more.

## Expected behavior
It should always get broadcasted name of BLE device in iOS version.

## Actual behavior
It gets broadcasted name of BLE device only if phone is unlocked on iOS. Otherwise it gets the name of a PC's native bluetooth device.

## Configuration

**Version:** 0.9.3

**Platform:** iOS

**Device:** iPhone 5s/iPhone 6s ...

smstuebe, 2016-11-23:
Can you update to the latest version and check again, please? Current version is 1.1.0.

xabre, 2016-11-23:
@KolesnikValentine Did you enable bluetooth background mode? iOS stops most BLE operations while in background and this mode is not enabled in your Info.plist.

KolesnikValentine, 2016-11-24:
@smstuebe Sorry, I've forgotten to change the app version in the configuration section, I'm using the latest version of an app.
@xabre I've added NSLocationAlwaysUsageDescription to the Info.plist, as well as a few background modes including Location updates, Uses BLE accessories and Background Fetch.

KolesnikValentine, 2016-11-24:
@xabre A little update: I was wrong, it shows the name of the PC which runs the bluetooth dongles, but not the bluetooth name of the PC. And this bug persists every time your app goes to background or your screen is locked.

mariusbloemhof, 2017-05-04:
This is not a bug. When IOS scans when your app is in the background, it does not provide the device information (such as name). Scanning in the background should be limited to devices with known UUID's or that advertises specific known Service UUID's