dotnet-bluetooth-le/dotnet-bluetooth-le #118: Reading a Characteristic whose value is null crashes the app

leelatoo, 2016-11-24:
Hi,

We try to access a characteristic in our BLE device. The characteristic's value is 'null'. So, when we try to access, we should get null right?
 Following is the piece of code we used.
`public async Task<byte[]> Read(ICharacteristic characteristic)
  {
            byte[] data = {};

			try
			{
				data = await characteristic.ReadAsync();
			}
			catch (InvalidOperationException ex)
			{
				System.Diagnostics.Debug.WriteLine("InvalidOperationException reading the characteristics value: " + ex.Message);
			}
			catch (CharacteristicReadException ex)
			{
				_System.Diagnostics.Debug.WriteLine("CharacteristicReadException reading the characteristics value: " + ex.Message);
			}
			catch (ArgumentNullException ex)
			{
				System.Diagnostics.Debug.WriteLine("Excpetion in reading the characteristic " + characteristic.Id);
			}
                        catch(Exception ex)
                        {
				System.Diagnostics.Debug.WriteLine(ex.Message);
                        }

                return data;
        }`

## Steps to reproduce

1.  Used the above mentioned to read a characteristic value (which is null in BLE device)

2. App crashes

## Expected behavior
`byte[] data = await characteristic.ReadAsync();`  - ReadAsync() method should return null.
So, `data`should have the value as `null`. Ideally, app should not crash here.

## Actual behavior
App crashes.
Exception handling is done for reading the characteristic. But, the control does not come under any of the catch blocks which we have used. It crashes in Plugin.BLE.iOS.Characteristic line 71.

### Crashlog

Nov 24 16:15:13 Audi-iPhone-5S RRCiOS[222] <Warning>: Characteristic.ReadAsync
Nov 24 16:15:27 Audi-iPhone-5S RRCiOS[222] <Warning>: 
	Unhandled Exception:
	System.ArgumentNullException: Value cannot be null.
	Parameter name: src
	  at (wrapper managed-to-native) System.Runtime.InteropServices.Marshal:copy_from_unmanaged (intptr,int,System.Array,int)
	  at System.Runtime.InteropServices.Marshal.Copy (System.IntPtr source, System.Byte[] destination, System.Int32 startIndex, System.Int32 length) [0x00000] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/_ios-build/Library/Frameworks/Xamarin.iOS.framework/Versions/git/src/mono/mcs/class/corlib/System.Runtime.InteropServices/Marshal.cs:152 
	  at Foundation.NSData.ToArray () [0x00012] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/Foundation/NSData.cs:50 
	  at Plugin.BLE.iOS.Characteristic.get_Value () [0x00000] in C:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.iOS\Characteristic.cs:25 
	  at Plugin.BLE.iOS.Characteristic+<>c__DisplayClass15_0.<ReadNativeAsync>b__2 (System.Object sender, CoreBluetooth.CBCharacteristicEventArgs args) [0x00051] in C:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.iOS\Characteristic.cs:71 
	  at CoreBluetooth.CBPeripheral+_CBPeripheralDelegate.UpdatedCharacterteristicValue (CoreBluetooth.CBPeripheral peripheral, CoreBluetooth.CBCharacteristic characteristic, Foundation.NSError error) [0x00015] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/build/ios/native/CoreBluetooth/CBPeripheral.g.cs:496 
	  at (wrapper managed-to-native) UIKit.UIApplication:UIApplicationMain (int,string[],intptr,intptr)
	  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/UIKit/UIApplication.cs:79 
	  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x00038] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/UIKit/UIApplication.cs:63 
	  at RRC.iOS.Application.Main (System.String[] args) [0x00008] in /Users/audiologymacbook/Documents/AudiApps/TFSWS/CoffeeWorkspace/Feat/Coffee_Dev/RCApps/Source/RRC.iOS/Main.cs:17 

We downloaded plugin's source code and used it to read that particular characteristic.
Below is the screenshot of crash.
[image]

## Configuration

**Version:** 9.3.1

**Platform:** iOS

**Device:** iPhone 5S

smstuebe, 2016-11-24:
I think the problem is not `ReadAsync()` it's more the getter if the `Value` property. If so, I'll fix that.

smstuebe, 2016-11-24:
@xabre I don't agree leelatos expectation of a `null` value. I'd return `new byte[0]` for empty values, because I could not find a state that expresses a null value (see BLE 4.2 spec page 556 §4.8 Characteristc read). I'd assume the same for write (=disallow null values).

What do you think?

xabre, 2016-11-25:
@smstuebe agreed

leelatoo, 2016-11-26:
@xabre returning `new byte[0]` for empty values will work for us.
In case of any other exceptions while reading/writing/subscribing, we want to handle the exceptions.

smstuebe, 2016-11-27:
available as beta2

leelatoo, 2016-11-28:
@smstuebe Thanks 👍  It works.

closed 2016-11-28 by smstuebe

MarcoGarciaV, 2019-04-27:
@leelatoo I have the same problem, how did you solve it? :/