dotnet-bluetooth-le/dotnet-bluetooth-le #119: Question on ValueUpdate

alberk8, 2016-11-27:
I need advise on how about to do the following

1) I have a Notify Characteristic with a XXNotifyCharacteristic_ValueUpdated event handler
2) I need to WriteAsync to run sequential order with the ValueUpdated completed prior to running  the ProcessData.

At the moment I have to put the Task.Delay statement otherwise the ProcessData method will execute prior to one or both the ValueUpdate are returned.  

For example 

```
await XXNotifyCharacteristic.WriteAsync( byte[]{81,45,1}));
await Task.Delay(400);
await XXNotifyCharacteristic.WriteAsync( byte[]{81,46,2}));
await Task.Delay(400);
ProcessData(value45Cmd, value46Cmd);

 private void XXNotifyCharacteristic_ValueUpdated(object sender, CharacteristicUpdatedEventArgs e)
        {
            if(e.Characteristic.Value[0] == 81 && e.Characteristic.Value[1] == 45)
            {
                value45Cmd = e.Characteristic.Value;
               }

            if (e.Characteristic.Value[0] == 81 && e.Characteristic.Value[1] == 46)
            {
                value46Cmd = e.Characteristic.Value;
            }
}
```

xabre, 2016-11-27:
Use a task completion source :) Here's a quick idea. Of course you should clean it up:

```
_tcs = new TaskCompletionSource<bool>();
await XXNotifyCharacteristic.WriteAsync( byte[]{81,45,1}));
await _tcs.Task();

_tcs = new TaskCompletionSource<bool>();
await XXNotifyCharacteristic.WriteAsync( byte[]{81,46,2}));
await tcs.Task();

ProcessData(value45Cmd, value46Cmd);

 private void XXNotifyCharacteristic_ValueUpdated(object sender, CharacteristicUpdatedEventArgs e)
        {
            if(e.Characteristic.Value[0] == 81 && e.Characteristic.Value[1] == 45)
            {
                value45Cmd = e.Characteristic.Value;
                _tcs?.TrySetResult(true);
               }

            if (e.Characteristic.Value[0] == 81 && e.Characteristic.Value[1] == 46)
            {
                value46Cmd = e.Characteristic.Value;
                _tcs?.TrySetResult(true);
            }
}
    
```

alberk8, 2016-11-27:
Thank you.  I learn something new today.

closed 2016-11-27 by smstuebe

alberk8, 2016-11-28:
Very sorry to reopen this.  How do I pass in a cancellation token to the TaskCompletionSource?  Since from my observation of my Bluetooth device it can play possum at times then the await _tcs.Task could wait forever.

xabre, 2016-11-30:
You can register it on you cancelationtoken. 
`cancellationToken.Register(() => _tcs?.TrySetCancelled(cancellationToken));`