dotnet-bluetooth-le/dotnet-bluetooth-le #121: App crashes accessing characteristics

leelatoo, 2016-12-06:
Hi,

In our app startup, we access a number of characteristics in a BLE device. Most of the times, accessing these characteristics works right out of the box. Randomly app crashes when it tries to access some characteristics. It does not crash frequently (1 out of 20 times).

## Steps to reproduce

1. Launch the app. app connects to BLE device and tries to access a few characteristics.
2. App crashes

## Expected behavior
App should not crash or it should throw exception to us

## Actual behavior
App crashed

### Crashlog

Error Stack Trace :
<Warning>: Warning: Unhandled exception: System.InvalidOperationException: Collection was modified; enumeration operation may not execute.

              at System.ThrowHelper.ThrowInvalidOperationException (System.ExceptionResource resource) <0x10040caf0 + 0x00024> in <a11ea1906afa4ff781670d3ba1c4a833#102d2e483761e0bb29e8adb6c07c8940>:0 

              at System.Collections.Generic.List`1+Enumerator[T].MoveNextRare () <0x100322d70 + 0x0002f> in <a11ea1906afa4ff781670d3ba1c4a833#102d2e483761e0bb29e8adb6c07c8940>:0 

              at System.Collections.Generic.List`1+Enumerator[T].MoveNext () <0x100322ca0 + 0x0009f> in <a11ea1906afa4ff781670d3ba1c4a833#102d2e483761e0bb29e8adb6c07c8940>:0 

              at System.Linq.Enumerable.FirstOrDefault[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] predicate) <0x100719ad0 + 0x000cf> in <45ddf04144214dd89fc1e7215a8a94aa#102d2e483761e0bb29e8adb6c07c8940>:0 

              at Plugin.BLE.Abstractions.ServiceBase+<GetCharacteristicAsync>d__12.MoveNext () <0x100ab86e0 + 0x0029f> in <0fdbf1b4ae3a48c4887b46bc9db47baa#102d2e483761e0bb29e8adb6c07c8940>:0 

            --- End of stack trace from previous location where exception was thrown ---

## Configuration

**Version:** 9.3.1

**Platform:** iOS

**Device:** iPhone 5S

xabre, 2016-12-06:
Seems like a characteristic was added to the collection while already searching for another.

xabre, 2016-12-17:
Relesed 1.2.0-beta3. Please let us know if it still happens.

closed 2016-12-17 by xabre

leelatoo, 2016-12-20:
Thanks for releasing a beta version with fix.
We tested the 1.2.0-beta3, it works.
As the mentioned crashes does not occur always, we will report if we find the crash again.

Thanks :+1: