dotnet-bluetooth-le/dotnet-bluetooth-le #123: Sampe App doesn't work

Maaarek, 2016-12-09:
Hi,

deployed your sample am on my phone but it doesn't start to detect any kind of devices .
Bluetooth is on.  

## Steps to reproduce

1.

2.

3.

## Expected behavior
Tell us what should happen

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version:** 0.9.3

**Platform:** iOS/Android/ ... (including version!!! e.g. Android 5.1 / i0S 10)

**Device:** HTC Sensation/ ...

smstuebe, 2016-12-11:
That's a **bit** too little information. What do you want us to do with the information you provided?

Maaarek, 2016-12-11:
It is now working very well.

closed 2016-12-11 by Maaarek

xabre, 2016-12-11:
You probably forgot to grant the location permissions (check logcat) ... or pull to refresh the list :P