dotnet-bluetooth-le/dotnet-bluetooth-le #124: Unable to subscribe for the BLE notifications in android

AnoopChatra, 2016-12-09:
Hi ,

   In android  I am unable to subscribe for the BLE notification for a particular characteristic though it supports notification.

Will there be any issue if I await the subscribe call ??
 

## Steps to reproduce

1.I have successfully connected to a BLE device using : 
CrossBluetoothLE.Current.Adapter.ConnectToDeviceAsync(deviceToBeConnected);

2. I have successfully read all the services and characteristics

3.I am trying to subscribe for the notification for a particular characteristic of a particular service, using :
 await characteristic.StartUpdatesAsync();

## Expected behavior
The app should successfully register for the notifications

## Actual behavior
The app is crashing 

### Crashlog
2-09 09:01:02.183 I/MonoDroid(25485): UNHANDLED EXCEPTION:
12-09 09:01:02.202 I/MonoDroid(25485): abc.xyz.BluetoothLeException: Exception occured in StartNotify()  ---> System.Exception: GATT: WRITE descriptor value failed
12-09 09:01:02.202 I/MonoDroid(25485):   at Plugin.BLE.Android.Descriptor.InternalWrite (System.Byte[] data) [0x0002c] in <7bece02839074605be3fd9668a0f909c>:0 
12-09 09:01:02.202 I/MonoDroid(25485):   at Plugin.BLE.Android.Descriptor+<>c__DisplayClass8_0.<WriteNativeAsync>b__0 () [0x00000] in <7bece02839074605be3fd9668a0f909c>:0 
12-09 09:01:02.202 I/MonoDroid(25485):   at Plugin.BLE.Abstractions.Utils.TaskBuilder+<FromEvent>d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext () [0x000b3] in <0fdbf1b4ae3a48c4887b46bc9db47baa>:0 
12-09 09:01:02.202 I/MonoDroid(25485): --- End of stack trace from previous location where exception was thrown ---
12-09 09:01:02.202 I/MonoDroid(25485):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/3819/c1d1c79c/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
12-09 09:01:02.202 I/MonoDroid(25485):   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00047] in /Users/builder/data/lanes/3819/c1d1c79c/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187 
12-09 09:01:02.202 I/MonoDroid(25485):   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in /Users/builder/data/lanes/3819/c1d1c79c/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156 
12-09 09:01:02.202 I/MonoDroid(25485):   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in /Users/builder/data/lanes/3819/c1d1c79c/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128 
12-09 09:01:02.202 I/MonoDroid(25485):   at System.Runtime.CompilerServices.TaskAwaiter.GetResult () [0x00000] in /Users/builder/data/lanes/3819/c1d1c79c/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:113 
12-09 09:01:02.202 I/MonoDroid(25485):   at Plugin.BLE.Android.Characteristic+<StartUpdatesNativeAsync>d__21.MoveNext () [0x002aa] in <7bece02839074605be3fd9668a0f909c>:0 

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version:** Plugin.BLE.1.2.0-beta2

**Platform:** Android 6.0

**Device:** LG Nexus 5

xabre, 2016-12-09:
Not sure if this is the cause of your problem, but can you try to call await characteristic.StartUpdatesAsync(); from the main thread? Can you paste some code which shows the sequence of your operations i.e. : UI interaction -> Connect -> start updates

AnoopChatra, 2016-12-09:
Hi ,

Thanks for the quick reply. I will send you the requested code snippet on
Monday morning.

Regards,
Anoop Chatra

sarkreth, 2016-12-14:
Are you sure that the descriptor is writeable?

AnoopChatra, 2016-12-15:
@xabre , I am making a call from main thread only.
@sarkreth , I am sure that descriptor is writable. I am able to subscribe for the notification when I make a call from iOS device.
I have found another interesting thing, subscription to the notification works fine in android as well If I put a delay of 10 seconds  just before making the "characteristic.StartUpdatesAsync();" call . Is it something to do with multi threading ??

xabre, 2016-12-16:
@AnoopChatra An interesting point. Can you show what other operation you are doing before calling start updates? Can you paste some sample code which shows the order of operations? I ask this because chaining operation sometimes causes weird timin issues on Android.

AnoopChatra, 2016-12-23:
@xabre  I have updated the library to Plugin.BLE.1.2.0-beta3
Now I am facing some new problems .  I am unable to write to a particular characteristics in android. I have attached a detailed doc about the issue. The doc also contains the info about  workflow of the app. 

[AndroidBLE.txt](https://github.com/xabre/xamarin-bluetooth-le/files/670493/AndroidBLE.txt)

Since Registering for the notification is in the later stages of my workflow, I could not test the old issue(crash in registering for the the  BLE notifications)

xabre, 2016-12-23:
How are you chaining these operations in your Pair method?

Are you doing something like this:

``
await Connect(d);
await GetService(device);

for some characteristic : c
{
    await StartNotify(c, ...)
}

for some data
await c.Write(data)

``

What I mean is that all BLE operations need to wait for the previous ones to complete (you should not start two BLE operations in parallel). Moreover, the BLE stack implementation on android has a limit to the number of characteristics you can subscribe to (I think 4... not sure).

My suggestion for debugging, comment out all your BLE calls except connect, and then start un-commenting them one by one until it stops working...then you will hopefully uncover why.

AnoopChatra, 2016-12-23:
Hi 

I am exactly following the approach how you have mentioned. 
Along with this I am also performing multiple read operations.

Now I am  able to pin point at a particular characteristic. Writing to it this characteristic is always throwing an exception.   Is there any reason behind this kind of strange behavior

xabre, 2016-12-23:
if you only have the connect code, discover services, characteristics and then try write (without any notifciations, read operation or other characteristic opertions), does it still fail?

xabre, 2017-01-19:
@AnoopChatra Is this still an issue for you?

AnoopChatra, 2017-01-19:
@xabre  I am not Sure.  
I am often getting "GATT 133" error while connecting to the BLE device. So I could not check the notification part properly.

xabre, 2019-08-09:
Outdated. Closing

closed 2019-08-09 by xabre

ferk0l, 2019-12-04:
I'm having the exact same issue. When not waiting, I get a "GATT: WRITE descriptor value failed" when trying to subscribe to a characteristic. 

I also occasionally get a GATT 133 error when trying to connect to my peripheral.

Are there any updates on this?