dotnet-bluetooth-le/dotnet-bluetooth-le #125: Question: How do I maximize WriteAsync()'s throughput?

sarkreth, 2016-12-15:
My application occasionally needs to write a lot of data over BLE (e.g. firmware updates). I've been doing that by just sending 20B in a loop, "await"ing every WriteAsync(), like this:
```
while (size > 20) //divide msg into chunks of max MTU
{
	Array.Copy(msg, msg.Length - size, chunk, 0, 20);
	await writeBuf.WriteAsync(chunk);
	size -= 20;
}
```
but I only get ~200B/s with this approach.

Reads are an order of magnitude faster, so I know there isn't a limitation in either of the devices' stacks. Is there some way to use the API that I'm not aware of?

## Configuration

**Version:** 1.1.0

**Platform:** Android 6.0

**Device:** Samsung Galaxy S7, prototype peripheral with a Nordic chip that I'm also programming

xabre, 2016-12-17:
On Android the implementation of writeAsync writes the data and waits for the gatt callback method (onCharacteristicWrite) to be called indicating the result of the operation. 

The characteristic write type does influence the transfer speeds. I.e. Best case is for  **WriteWithoutResponse**. 

This is a great question. I had to deal with a similar issue in a previous project for exactly the same use case: firmware update. The best we got was similar to your results (unfortunately limited by the peripheral implementation where we had to wait for a notification after each package was written). 

I still want to find some time to implement a ble "server" package and then extend the sample app to act as a peripherial and try to optimize and fine tune large data transfers.

sarkreth, 2016-12-28:
Ah. I used a BLE sniffer, and indeed, I wasn't using WriteWithoutResponse, even though the peripheral supports it. For anyone else looking at this, after the characteristic has been gotten, use
myCharacteristic.WriteType = Plugin.BLE.Abstractions.CharacteristicWriteType.WithoutResponse;
For my device, throughput is now ~2kB/s

closed 2016-12-28 by sarkreth