dotnet-bluetooth-le/dotnet-bluetooth-le #126: Accessing characteristic crashes app

leelatoo, 2016-12-16:
## Steps to reproduce

1. Connect to BLE device

2. Access services and characteristics

3. App crahses

## Expected behavior
In case of any exception in accessing services/characteristics, plugin should throw exception to app.
App should not crash.

## Actual behavior
App crashes

### Crashlog

at System.Linq.Enumerable.Select[TSource,TResult] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] selector) [0x00010] in <45ddf04144214dd89fc1e7215a8a94aa>:0 
	  at Plugin.BLE.iOS.Service+<>c__DisplayClass7_0.<GetCharacteristicsNativeAsync>b__0 (System.Object sender, CoreBluetooth.CBServiceEventArgs args) [0x0002e] in <8d1743147a5640d6a3a02f4ecf9ee65c>:0 
	  at (wrapper delegate-invoke) System.EventHandler`1[CoreBluetooth.CBServiceEventArgs]:invoke_void_object_TEventArgs (object,CoreBluetooth.CBServiceEventArgs)
	  at CoreBluetooth.CBPeripheral+_CBPeripheralDelegate.DiscoveredCharacteristic (CoreBluetooth.CBPeripheral peripheral, CoreBluetooth.CBService service, Foundation.NSError error) [0x00015] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/build/ios/native/CoreBluetooth/CBPeripheral.g.cs:401 
	  at (wrapper managed-to-native) UIKit.UIApplication:UIApplicationMain (int,string[],intptr,intptr)
	  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/UIKit/UIApplication.cs:79 
	  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x00038] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/UIKit/UIApplication.cs:63 
	  at Application.Main (System.String[] args) [0x00008] in Main.cs:17 
Dec 16 11:14:33 iPhone RRCiOS[307] <Warning>: Unhandled managed exception:
	Value cannot be null.
	Parameter name: source (System.ArgumentNullException)
	  at System.Linq.Enumerable.Select[TSource,TResult] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] selector) [0x00010] in <45ddf04144214dd89fc1e7215a8a94aa>:0 
	  at Plugin.BLE.iOS.Service+<>c__DisplayClass7_0.<GetCharacteristicsNativeAsync>b__0 (System.Object sender, CoreBluetooth.CBServiceEventArgs args) [0x0002e] in <8d1743147a5640d6a3a02f4ecf9ee65c>:0 
	  at (wrapper delegate-invoke) System.EventHandler`1[CoreBluetooth.CBServiceEventArgs]:invoke_void_object_TEventArgs (object,CoreBluetooth.CBServiceEventArgs)
	  at CoreBluetooth.CBPeripheral+_CBPeripheralDelegate.DiscoveredCharacteristic (CoreBluetooth.CBPeripheral peripheral, CoreBluetooth.CBService service, Foundation.NSError error) [0x00015] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/build/ios/native/CoreBluetooth/CBPeripheral.g.cs:401 
	  at (wrapper managed-to-native) UIKit.UIApplication:UIApplicationMain (int,string[],intptr,intptr)
	  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/UIKit/UIApplication.cs:79 
	  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x00038] in /Users/builder/data/lanes/3818/3983064a/source/xamarin-macios/src/UIKit/UIApplication.cs:63 
	  at Application.Main (System.String[] args) [0x00008] in Main.cs:17 

## Configuration

**Version:** 9.3.1

**Platform:** iOS

**Device:** iPhone 5S

leelatoo, 2016-12-16:
Can you please add exception handling and give us a build?

Thanks in advance

xabre, 2016-12-17:
Relesed 1.2.0-beta3. Please let us know if it still happens.

closed 2016-12-17 by xabre

leelatoo, 2016-12-20:
Thanks for releasing a beta version with fix. 
We tested the 1.2.0-beta3, it works.
As the mentioned crashes does not occur always, we will report if we find the crash again.

  Thanks :+1: