dotnet-bluetooth-le/dotnet-bluetooth-le #128: Writing/reading >20 bytes

gaminn, 2016-12-19:
Hi,
is reading/writing >20 bytes from/to characteristic supported with this plugin? If so, please share how to do it?

xabre, 2016-12-23:
@gaminn Not yet :) 

We have an implementation idea (but we need more info) but nothing in the plugin yet:
- On android: use the RequestMtu call/callback 
- On iOS: not sure? open question?

Cheesebaron, 2016-12-23:
It seems to work fine for me? I just provide a byte array with the data and it eats it no problem. On Android I have had it eat around 450 bytes no problem.

sarkreth, 2016-12-28:
@Cheesebaron Are you sure the byte array makes it to the peripheral as expected? Also, what device are you using?

Cheesebaron, 2016-12-28:
Pretty sure it does. Have tried this on both iPhone and Android.

I can test again when I am back at work.

martinwiboe, 2017-01-05:
Can confirm that it only works for me if I call `RequestMtu` with a sufficiently large value prior to writing. I'm using a OnePlus 3 with Nougat.

xabre, 2017-01-06:
@martinwiboe Did you also try it on iOS?

martinwiboe, 2017-01-06:
@xabre No, not using your library, unfortunately. We do have a native iOS app that uses BLE to talk to the same product, and that works correctly without any adjustment to MTU etc.

martinwiboe, 2017-01-11:
@Cheesebaron have you had time to test this? :)

Dragon160, 2017-04-21:
We did an airtrace and discovered that iOS automatically calls an Exchange MTU request after establishing the connection. 
Android does not show this behaviour. I think the "requestMTU(xxx)" call needs to be done by the application itself.

@xabre Any plans to adopt this? :-)

closed 2019-08-11 by xabre