dotnet-bluetooth-le/dotnet-bluetooth-le #129: Add CBCentralInitOptions support

flyingxu, 2016-12-27:
On iOS, currently `CBCentralManager` is initialized by 

`_centralManager = new CBCentralManager(DispatchQueue.CurrentQueue);`

https://github.com/xabre/xamarin-bluetooth-le/blob/e20941a7b9aeecec8d6d865652c7f76eae4bcc5b/Source/Plugin.BLE.iOS/BleImplementation.cs#L22

There is no way to pass in `CBCentralInitOptions` parameters. by `CBCentralInitOptions`, [system should display a warning dialog to the user if Bluetooth is powered off ](https://developer.apple.com/reference/corebluetooth/cbcentralmanageroptionshowpoweralertkey), then the user can be redirected to iOS setting, which is convenient for the user.

bwenninger, 2018-01-05:
Is there a plan to add the ability to pass in options to the CBCentralManager constructor?  I'm looking for a way to not have iOS alert the user that Bluetooth is not turned on and the CBCentralManagerOptionShowPowerAlertKey option would allow this but I can't currently pass that key in.

xabre, 2019-08-09:
Both **RestoreIdentifier** and **ShowPowerAlert** are supported now

PR #358 (2019-07-14): Added an option to ShowPowerAlert for iOS.
Fixes #129 

User can decide to enable or not the option `ShowPowerAlert` on iOS. When this option is enable, a message will be displayed to the user to turn On the bluetooth in the Settings when in airplane mode and bluetooth is Off. In our case, this message is annoying for our users.

Run the following within `AppDelegate.FinishedLaunching` before any other usage of the plugin: `BleImplementation.ShowPowerAlert(false);`.

Default value of `ShowPowerAlert` is true. The `ShowPowertAlert` will be display even if there is no options provided to `CBCent [... cut]
closed 2019-08-09 by xabre