dotnet-bluetooth-le/dotnet-bluetooth-le #130: GUID differents in iOS vs Android

Graveen, 2016-12-28:
I just went to see Android GUID are not matching iOS GUID.
I can filter on the Device.Name, but can't definitively do this on Device.Id
Perhaps i misread what must 

## Steps to reproduce

(snip)
BtAdapter.DeviceDiscovered += OnBtDeviceFound;
await BtAdapter.StartScanningForDevicesAsync();

private static void OnBtDeviceFound(object sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e)
{
      string ret =  $"Name:{e.device.Name} with Id:{e.Device.Id}";
}

## Expected behavior
I expect iOS and Android ID matching, as they are the hardware BT adress

## Actual behavior
Instead, this is different. 
Id on Android is 000000+ 6 bytes hw address
Id on iOS is full GUID unrelated to hw address

## Configuration

**Version:** 1.1.0

**Platform:** iOS/10.2 - Android 6.01

**Device:** all devices i own (different Android devices)

xabre, 2016-12-29:
Each iOS device assigns a GUID to a known BLE peripheral which has nothing to do with the it's mac (unfortunately). This means that the same Peripherial will have different 'id' s on all iPhones. This is a limitation of the apple implementation as far as I know. If you can find a way to get the hw address of a CB Peripherial please share it:)

Graveen, 2016-12-29:
Thank you Xabre for the clarification. I'll let you know if i find something relevant :)

xabre, 2017-02-18:
@Graveen any finidngs?

Graveen, 2017-02-20:
@xabre no sadly. After some research, i had to deal with this limitation.

closed 2019-08-09 by xabre