dotnet-bluetooth-le/dotnet-bluetooth-le #131: Directly using the source code in my application?

sarkreth, 2016-12-28:
I need to modify the Plugin for my application, namely adding a RequestMtu() function. So, I tried migrating my code to a copy of the solution of Plugin's source, but I end up with a runtime exception during initialization. I am new to C# and Xamarin, so if the fix is obvious, I'd be happy with a reference to remedial text.

## Steps to reproduce

1. Create a "normal" project, importing the library via NuGet.

2. Download the library's source, copy-paste the projects from Step 1 into the Source folder, add them in the solution, remove Plugin.BLE from the NuGet Packages.

3. Deploy the app to an Android device.

## Expected behavior
The app runs as it did before.

## Actual behavior
During initializtion in the Portable project code, `var ble = CrossBluetoothLE.Current;` causes the exception:
"This functionality is not implemented in the portable version of this assembly.  You should reference the NuGet package from your main application project in order to reference the platform-specific implementation."
I do notice that "PORTABLE" is defined for CrossBluetoothLE.cs, so the assignment seems "doomed to fail" as I have it set up. Am I not supposed to use that line in the Portable project, or do I have some incorrect references, or what?

## Configuration

**Version:** 1.1.0, Vanilla usage

**Platform:** Android 6.0

**Device:** Samsung Galaxy S7

sarkreth, 2017-01-20:
I fixed it. Issue should be closed. As I suspected, my issues stemmed from the fact that I'm a noob. There was a lot I didn't understand, so I would also consider this off-topic.

closed 2017-01-21 by smstuebe

MTigra, 2018-04-14:
so what was the solution?

sarkreth, 2018-05-27:
This was quite some time ago, so I don't remember exactly what I did. I remember the issue stemming from the fact that I wasn't using the MVVM projects. I was missing a couple of files that attached cross-platform and platform-specific, so I made new ones that were just a couple of lines long and everything worked.