dotnet-bluetooth-le/dotnet-bluetooth-le #132: Weird peripheral name + connection error = crash

sarkreth, 2017-01-03:
## Steps to reproduce

1. Make a peripheral with a "weird" name (i.e. using unprintable characters).

2. Program the peripheral to interrupt and drop its own connection procedure.

3. Attempt a connection on the Central.

## Expected behavior
I get this error message
[image]

## Actual behavior
"Unhandled Exception:
System.FormatException: Input string was not in a correct format."
is thrown by `TraceImplementation?.Invoke(format, args);` in Trace.cs

## Configuration

**Version:** 1.1.0

**Platform:** Android 6.0

**Device:** Samsung Galaxy S7

xabre, 2017-01-03:
Thanks for pointing it out. Currently the standard Trace implementation is Console.WriteLine. This implementation handles all the output messages from the plugin. You can override it with your own implementation like this:

Somewhere in your app init add: 
`
Plugin.BLE.Abstractions.Trace.TraceImplementation = // anything which is a Action<string, object[]> 
`

smstuebe, 2017-01-08:
Will be fixed with the next release.

closed 2017-01-08 by smstuebe