dotnet-bluetooth-le/dotnet-bluetooth-le #133: Does not scan for Beacons

keithmak, 2017-01-04:
I tried this package but it doesn't scan for iBeacon, it succeeded scanning a watch.

sarkreth, 2017-01-04:
I refer you to https://github.com/xabre/xamarin-bluetooth-le/issues/104
Did you enable location permissions?

Also, what platform, platform version, and plugin version are you using?

keithmak, 2017-01-05:
It is running Android 4.4.2 KOT49H, MIUI Global 8.0 (KHDMIDG), plugin version 1.1.0

I added these in Manifest but still not seeing beacons
Additionally I enabled location permission in my Android Setting

<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.BLUETOOTH" />
  <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
  <uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
  <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

keithmak, 2017-01-05:
(SUCCEED)

Thank you sarkreth. Turned out my code contains other bugs, thats why it is not recognizing the beacons even after I added the Location Permission as you have instructed

closed 2018-06-23 by flyingxu