dotnet-bluetooth-le/dotnet-bluetooth-le #135: Bluetooth turned on fired twice

leelatoo, 2017-01-09:
Hi, 

We want to handle device's bluetooth turned off and turned on event in our app.
Registered for _bluetoothLe.StateChanged event.

## Steps to reproduce

1. Register _bluetoothLe.StateChanged event.

2. Turn off bluetooth in device. StateChanged event is fired.

3. Turn on bluetooth in device. StateChanged event is fired twice.

## Expected behavior
When device's Bluetooth is turned on, StateChanged event should be fired only once.

## Actual behavior
When device's Bluetooth is turned on, StateChaged event is fired twice. Because of this issue, bluetooth state change handling is not working properly.

## Configuration

**Version:** 9.1

**Platform:** iOS

**Device:** iPhone 6

smstuebe, 2017-01-09:
Your expectation is wrong. The Ble state is not dual.

See: https://github.com/xabre/xamarin-bluetooth-le/blob/master/Source/Plugin.BLE.Abstractions/Contracts/BluetoothState.cs

On Android for example you will have the following transitions when you are turning on Bluetooth: `Off -> TurningOn (optional) -> On`
On iOS there is no `TurningOn` state counterpart, but the iOS Api can notify an other state that is not mapable directly to `TurningOn`. That's why you get the transitions `Off -> Unknown (optional) -> On`

What is printing in your case? 

```
private void OnStateChanged(object sender, BluetoothStateChangedArgs e)
{
    Debug.WriteLine("{0} -> {1}", e.OldState, e.NewState);
}
```

leelatoo, 2017-01-13:
@smstuebe Thanks for the information.

Here is the code and behavior of Bluetooth state changed event which i have got.. 
After bluetooth state changed event is fired, we print oldstate and newstate in console.

We turn off iPhone's bluetooth:
Log:
2017-01-13 12:11:02.278 BleApps : BleAdapter : BluetoothLeStateChanged event fired
2017-01-13 12:11:02.278 BleApps : BleAdapter : oldstate : On new state : Off
2017-01-13 12:11:02.295 BleApps UpdatedState: PoweredOff

We turn on iPhone's bluetooth immediately:
Log:
2017-01-13 12:11:08.129  BleApps : BleAdapter : BluetoothLeStateChanged event fired
2017-01-13 12:11:08.149 BleApps : BleAdapter : oldstate : Off new state : On
2017-01-13 12:11:13.891 BleApps : BleAdapter : BluetoothLeStateChanged event fired
2017-01-13 12:11:13.896 BleApps : BleAdapter : oldstate : Off new state : On
Thread finished: <Thread Pool> #15

I think the event is fired twice. Can you please analyse the logs and help us?

leelatoo, 2017-01-17:
@smstuebe 

And, we have noticed another issue in Plugin.BLE. When app tries to connect to a ble device, DeviceConnected event gets fired once. That is absolutely fine.

When the ble device gets disconnected, the DeviceDisconnected event gets fired multiple times.
When the ble device gets connected again, the DeviceConnected event gets fired multiple times.

smstuebe, 2017-01-20:
Hi,

I could not reproduce your issue, yet :( I'll have a closer look, soon. 
For the DeviceConnected issue, please create a new issue.

smstuebe, 2017-01-28:
@leelatoo  Can you provide a demo project, please? Just create a github repository, that I can clone and investigate.