dotnet-bluetooth-le/dotnet-bluetooth-le #136: Scan Low Latency Setting

alberk8, 2017-01-11:
Hi,

Is it possible to add support for SCAN_MODE_LOW_LATENCY  for Android 5.x and above ?

https://developer.android.com/reference/android/bluetooth/le/ScanSettings.html#SCAN_MODE_LOW_LATENCY

smstuebe, 2017-01-11:
Yes it is possible :) Adding it as feature request.

alberk8, 2017-01-11:
Thanks.  Since we are on the roll,  please include the "MATCH" series.   :)

smstuebe, 2017-01-11:
> please include the "MATCH" series.

What? I don't get it.

alberk8, 2017-01-11:
What I mean is since you are going to look at adding support for SCAN_MODE_LOW_LATENCY , it would be wonderful to add the MATCH_* scan setting.  Thank you.

smstuebe, 2017-01-11:
Hmm I had a quick look at MATCH_ stuff. It seems, that it is just for filtering. We abstracted the filtering.
You can use `StartScanningForDevicesAsync` with a filter callback.

```
adapter.StartScanningForDevicesAsync(device => device.Name.Contains("foo"));
```

https://github.com/xabre/xamarin-bluetooth-le/blob/master/Source/Plugin.BLE.Abstractions/Extensions/AdapterExtenstion.cs#L45

Or does it provide more?

alberk8, 2017-01-13:
Hi,
I came across addition information Android BlE.  It might be of some help.  Thank you.

https://devzone.nordicsemi.com/attachment/7dc38a5edf15efb8c328b13544a4ac94

Cheesebaron, 2017-04-04:
> We need to investigate if there are similar options on iOS.

There aren't similar controls on iOS.

smstuebe, 2017-04-04:
Implemented in 1.2.2.

    adapter.ScanMode = ScanMode.LowLatency;

janusw, 2022-07-01:
> What I mean is since you are going to look at adding support for SCAN_MODE_LOW_LATENCY , it would be wonderful to add the MATCH_MODE_STICKY, MATCH_NUM_FEW_ADVERTISEMENT series scan setting. Thank you.

The match mode has been added in #607. Therefore I'm finally closing this issue.

closed 2022-07-01 by janusw