dotnet-bluetooth-le/dotnet-bluetooth-le #138: [Android] DisconnectDeviceAsync does not disconnect Bluetooth connection

horacehylee, 2017-01-11:
I connected the device using` ConnectToKnownDeviceAsync(device_guid)` with stored device_guid.
After some operation, I use `await DisconnectDeviceAsync(IDevice device)` to disconnect the device. 
The IAdapter says there are no any connected device. However, using tools like "nRF Connect for Mobile", the device is still connected. There should be a bug for android to disconnect with the device.

`<package id="Plugin.BLE" version="1.1.0" targetFramework="monoandroid70" />`

OS: Android
Version: 1.1.0 (Plugin.BLE)

xabre, 2017-01-19:
Does it happen every time? Can you give us some more information please?

D-Bullock, 2017-02-10:
Hi,

I'm getting this error also. 

The easiest way to replicate the error is as following:
`Adapter.DeviceConnected += async (s,e)=>{ await DisconnectDeviceAsync(e.Device);}'

xabre, 2017-02-28:
@horacehylee @D-Bullock 
What is the exact error message?
I have tried your code in our sample app but unfortunately I can't reproduce it ... Maybe add a larger code sample?

flyingxu, 2017-03-01:
I'm having the save problem since v1.1 to v1.2 on iOS. I'm not sure if it's related to c# garbage collection or some reference to native resources.

My issue is like:

When I call `DisconnectDeviceAsync`, the device is not disconnected (my ble device will auto advertise when disconnected, it's reliable and I verified it with other tools). But without any other function call, I just wait for several seconds, the phone is disconnected with my ble device. Very strange.

One more clue is that if I double press the home button and kill the app completely, the ble device is disconnected right away.

I tested on several iPhones, it seems that it's worse (can't disconnect) on iPhone 4s than iPhone 5.

sarkreth, 2017-06-26:
I was having the same problem, but I realized it was due to some debug code I added. This caused the disconnect method to not work:
```
myService = new myServiceClass(discCharacteristics);
myServiceCopy = myService;
myServiceCopy.EnableCharNotifications();
```
Removing myServiceCopy and just using myService for all of my characteristic operations fixed the issue.

luqmanhussain, 2017-12-27:
I also have the same issue as OP.
Its not disconnecting.  

So for instance after    await _adapter.DisconnectDeviceAsync(device);
If i perform  _adapter.GetSystemConnectedOrPairedDevices(ServiceGuidsList);
the device will still show in the results but shouldnt.

veletron2, 2018-02-07:
Just to say,I have the same issue here although on mine it disconnects just fine on Android (7.1), but the disconnect does nothing on iPhone 7 (ios 11.1)

mullerch, 2018-02-14:
I tested the demo application to know if the library is suitable for my project. I also have this problem and could identify some scenarios :

Android :
If the device and the smartphone are paired, asking a disconnection from the smartphone side will never disconnect on the device side (I don't know for the smartphone side).

Here are two other scenarios where the behavior is not as expected, but maybe this situation should never happen.

Android :
If the device is connected without pairing BUT the device is paired on the smartphone side (due to precedent pairing and profile change on the device side), it will never disconnect on the device side when disconnection is asked from the smartphone.

iOS :
If the device is connected without pairing BUT the device is paired on the smartphone side (due to precedent pairing and profile change on the device side), it will fail to connect when connection is asked from the smartphone.

Also, I have another one that I can now reproduce every time but I think it did not happen before I have paired the device and then "forgot" it on the smartphone side and cleared pairing on the device side.

Android :
Asking a disconnection from the smartphone side will never disconnect on the device side .

The first and last problem are very critical because it makes the library not usable at production level.

My test devices :
- Android 8.1.0 OPM1.171019.016 // Google Pixel PVT (the first, small one)
- iOS 11.1.2 15B202 // iPhone MG4H2PK/A

If you need additional details, do not hesitate.

mullerch, 2018-02-14:
I was NOT able to reproduce any of the described problems on Android with the following devices :
- Android 4.4.2 // Samsung SM-T230
- Android 7.0 // Samsung SM-T580

andrew-nemtsev, 2018-02-22:
Have the same issue with DisconnectDeviceAsync() at Huawei Honor 8Lite and Lenovo P2 (Android 7.0) with Plugin.BLE v1.3.0.
After DisconnectDeviceAsync() is awaited IDevice is moved to Plugin.BLE.Abstractions.DeviceState.Limited state, but according to nRF Connect BLE device is still at Connected state.

rileybwesley, 2018-04-03:
I am seeing the same results as @andrew-nemtsev . Any movement on this?

scoutret, 2020-11-09:
Hello, DisconnectDeviceAsync

MOI-MThomas, 2023-05-17:
I'm having this issue as well on Android.  I have a disconnect button in which I call `DisconnectAsync()` and half the time the device I'm connected to the connection is still there.  But sometimes it does work.  Not sure, I've even tried checking if `_device.State == DeviceState.Connected` and it comes back true, but it's still connected.

`	private async void OnDisconnectButtonClicked(object sender, EventArgs e)
	{
		var adapter = CrossBluetoothLE.Current.Adapter;
		await adapter.DisconnectDeviceAsync(_device);

		await Navigation.PopAsync();
	}`

derek-willis, 2023-07-10:
Samsung Note 8, Android 9, using .NET Maui, using Plugin.BLE v3.0.0-beta4 (vanilla)

I have this problem but only immediately after BONDING. If I call await _adapter.DisconnectDeviceAsync(device); it does not disconnect and after a timout period of about 20 seconds an exception is raised saying the task was cancelled.

If I close the app and reopen it, I am able to connect to the BONDED device and also disconnect from it successfully. It seems there is an issue immediately after bonding. This really breaks our product design !