dotnet-bluetooth-le/dotnet-bluetooth-le #140: Plugin not able to find BLE device at time of scan for Android 6 for Sansung Galaxy Tab S2

ravikanasagra1, 2017-01-12:
## Steps to reproduce

1. We have implemented Plugin BLE in our xamarin forms project. It is working fine with IOS App but has some issue with Android.

2. When we scan for BLE devices, it does not find device and times out. We use Samsung Galaxy S2 Tab bought in India. However, we have same tablet bought in USA and tested in USA, where same application works and finds device on scanning. We have just installed plugin from Nuget Package and does not integrated permission for Android M as mentioned. Still it worked in USA bought tablet but not in India bought tablet.

## Expected behavior
Plugin should scan BLE device with Indian Samsung Galaxy Tab S2.

## Actual behavior
Tab with Android 6 bought in India is not scanning devices.

**Version of the Plugin:** => 1.0.0 

**Platform:** Android 6.0.1

**Device:** Samsung Galaxy Tab S2

smstuebe, 2017-01-12:
Do other BLE Scanner Apps like nRF Connect for Mobile ? https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en

ravikanasagra1, 2017-01-13:
I checked with other BLEChecker app and it detects BLE device with India bought tablet.

smstuebe, 2017-01-13:
Can you provide some logs of the plugin?

ravikanasagra1, 2017-01-13:
Kindly suggest from where did I get logs in application.

xabre, 2017-01-13:
Application Output in XS or Output in VS or LogCat

ravikanasagra1, 2017-01-13:
Here is Output from Visual Studio.

01-13 16:04:23.770 D/Mono    ( 3108): [0x7e751930] worker finishing
Thread finished: <Thread Pool> #26
The thread 'Unknown' (0x1a) has exited with code 0 (0x0).
01-13 16:04:27.620 D/BluetoothAdapter( 3108): onBluetoothStateChange: up=true
01-13 16:04:27.620 D/BluetoothAdapter( 3108): onBluetoothStateChange: Bluetooth is on
01-13 16:04:29.130 V/ActivityThread( 3108): updateVisibility : ActivityRecord{d8f9383 token=android.os.BinderProxy@83d09ae {com.monsanto.ConnectIN/md5e85e373cd3b4f7e338cf2fa3349b8c60.MainActivity}} show : true
01-13 16:04:34.530 I/Timeline( 3108): Timeline: Activity_idle id: android.os.BinderProxy@83d09ae time:10587366
01-13 16:04:38.160 D/ViewRootImpl( 3108): ViewPostImeInputStage processPointer 0
01-13 16:04:38.300 D/ViewRootImpl( 3108): ViewPostImeInputStage processPointer 1
01-13 16:04:38.590 D/Mono    ( 3108): Assembly Ref addref InventoryManager[0xadd509a0] -> Plugin.BLE.Abstractions[0xadd4ffe0]: 3
01-13 16:04:38.590 D/Mono    ( 3108): Assembly Ref addref InventoryManager[0xadd509a0] -> Plugin.BLE[0xadd50040]: 2
01-13 16:04:38.630 D/Mono    ( 3108): Unloading image System.Collections.dll [0x85902000].
01-13 16:04:38.630 D/Mono    ( 3108): Image addref System.Collections[0x7a697140] -> System.Collections.dll[0xae897c00]: 12
01-13 16:04:38.630 D/Mono    ( 3108): Config attempting to parse: 'System.Collections.dll.config'.
01-13 16:04:38.630 D/Mono    ( 3108): Config attempting to parse: '/Users/builder/data/lanes/3819/96c7ba6c/source/monodroid/builds/install/mono-armv7/etc/mono/assemblies/System.Collections/System.Collections.config'.
01-13 16:04:38.630 D/Mono    ( 3108): Assembly Ref addref Plugin.BLE.Abstractions[0xadd4ffe0] -> System.Collections[0xadd50c40]: 10
01-13 16:04:38.655 D/Mono    ( 3108): Unloading image System.Threading.dll [0x85902000].
01-13 16:04:38.655 D/Mono    ( 3108): Image addref System.Threading[0x6d803200] -> System.Threading.dll[0xae8bdd00]: 10
01-13 16:04:38.655 D/Mono    ( 3108): Config attempting to parse: 'System.Threading.dll.config'.
01-13 16:04:38.655 D/Mono    ( 3108): Config attempting to parse: '/Users/builder/data/lanes/3819/96c7ba6c/source/monodroid/builds/install/mono-armv7/etc/mono/assemblies/System.Threading/System.Threading.config'.
01-13 16:04:38.655 D/Mono    ( 3108): Assembly Ref addref Plugin.BLE.Abstractions[0xadd4ffe0] -> System.Threading[0xb4265ee0]: 8
01-13 16:04:38.665 D/Mono    ( 3108): Assembly Ref addref Plugin.BLE[0xadd50040] -> System.Core[0xae8d39c0]: 12
01-13 16:04:38.680 D/Mono    ( 3108): DllImport attempting to load: '/system/lib/liblog.so'.
01-13 16:04:38.685 D/Mono    ( 3108): DllImport loaded library '/system/lib/liblog.so'.
01-13 16:04:38.685 D/Mono    ( 3108): DllImport searching in: '/system/lib/liblog.so' ('/system/lib/liblog.so').
01-13 16:04:38.685 D/Mono    ( 3108): Searching for '__android_log_print'.
01-13 16:04:38.685 D/Mono    ( 3108): Probing '__android_log_print'.
01-13 16:04:38.685 D/Mono    ( 3108): Found as '__android_log_print'.
01-13 16:04:38.685 I/mono-stdout( 3108): Adapter >=21: Starting a scan for devices with service Id 713d0000-503e-4c75-ba94-3148f18d941e.
Adapter >=21: Starting a scan for devices with service Id 713d0000-503e-4c75-ba94-3148f18d941e.
01-13 16:04:38.700 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:38.700 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:38.705 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:38.705 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:38.715 D/BluetoothLeScanner( 3108): Start Scan
01-13 16:04:38.715 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:38.720 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:38.720 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:38.720 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:38.765 D/BluetoothLeScanner( 3108): onClientRegistered() - status=0 clientIf=7
01-13 16:04:38.785 D/AbsListView( 3108): viewType is heaer or footer
01-13 16:04:38.885 D/AbsListView( 3108): viewType is heaer or footer
01-13 16:04:38.885 D/AbsListView( 3108): viewType is heaer or footer
01-13 16:04:38.895 D/AbsListView( 3108): viewType is heaer or footer
Thread started: <Thread Pool> #29
Adapter: Scan timeout has elapsed.
01-13 16:04:48.800 I/mono-stdout( 3108): Adapter: Scan timeout has elapsed.
Adapter: Stopping the scan for devices.
01-13 16:04:48.805 I/mono-stdout( 3108): Adapter: Stopping the scan for devices.
Adapter >= 21: Stopping the scan for devices.
01-13 16:04:48.810 I/mono-stdout( 3108): Adapter >= 21: Stopping the scan for devices.
01-13 16:04:48.810 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:48.810 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:48.815 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:48.815 D/BluetoothAdapter( 3108): STATE_ON
01-13 16:04:48.815 D/BluetoothLeScanner( 3108): Stop Scan
01-13 16:04:50.155 D/ViewRootImpl( 3108): #1 mView = com.android.internal.policy.PhoneWindow$DecorView{4a1d935 V.E...... R.....I. 0,0-0,0}
01-13 16:04:50.165 D/AbsListView( 3108): viewType is heaer or footer
01-13 16:04:50.270 D/AbsListView( 3108): viewType is heaer or footer
01-13 16:04:50.270 D/AbsListView( 3108): viewType is heaer or footer
01-13 16:04:50.280 D/AbsListView( 3108): viewType is heaer or footer
01-13 16:04:50.360 D/mali_winsys( 3108): new_window_surface returns 0x3000,  [1024x402]-format:1
01-13 16:04:50.395 D/ViewRootImpl( 3108): MSG_RESIZED_REPORT: ci=Rect(0, 0 - 0, 0) vi=Rect(0, 0 - 0, 0) or=2

ravikanasagra1, 2017-01-16:
I found above shared logs from output window in my Visual Studio 2015. Kindly let me know if you need any other information for troubleshooting.

xabre, 2017-01-19:
@ravikanasagra1 
I see you are using the service guid to filter the devices, some andorid ble implementations are having trouble with this, can you try scanning without any service guid and see if the device is discovered?

ravikanasagra1, 2017-01-19:
I tried in past also without GUID also but still it did not work.

rusty21, 2017-03-02:
I am experiencing this issue too on my Galaxy J3 running android 6. Has this been looked into more? I can confirm that a 3rd party BLE scanner can find devices, but Plugin.Ble finds no devices.

Denaindra, 2020-11-17:
HI @xabre , @ravikanasagra1 , @rusty21 

Thanks for contributing to this issue. Do you have any chance to get rid of this issue. I also facing the same issue from my development enviromnt. Im unable to scan any mobile and tab devices. but third party BLE enabled device could scan. 

If you have any solution please share it with me ... 

Cheers -