dotnet-bluetooth-le/dotnet-bluetooth-le #143: StartScanningForDevicesAsync()

BorisFR, 2017-01-18:
I have random success using StartScanningForDevicesAsync().

## Steps to reproduce
1. connect to a device
2. search for services

## Expected behavior
Obtain a list of available services

## Actual behavior
Sometime it is working but very often nothing happen and then the device is disconnected.

### trace when it is working
```
2017-01-18 15:57:25.065 BatKeeper.iOS[2936:567501] Adapter: Stopping the scan for devices.
2017-01-18 15:57:25.066 BatKeeper.iOS[2936:567501] Adapter: Scan was cancelled.
** ConnectToDevice - trying to connect to MI Band 2...
2017-01-18 15:57:26.861 BatKeeper.iOS[2936:567501] ConnectedPeripherial: MI Band 2
** Device connected 1a00527a-c5b1-492f-adfd-e074e99a56b6: MI Band 2 / Connected / <CBPeripheral: 0x1702e0d80, identifier = 1A00527A-C5B1-492F-ADFD-E074E99A56B6, name = MI Band 2, state = connected> :)
> Connected, searching services
*** Searching services
*** SearchBleServices - async...
2017-01-18 15:57:26.880 BatKeeper.iOS[2936:567501] ConnectToDeviceAsync Connected: 1a00527a-c5b1-492f-adfd-e074e99a56b6 MI Band 2
2017-01-18 15:57:28.939 BatKeeper.iOS[2936:567501] Device.Discovered Service: <CBService: 0x17026b880, isPrimary = YES, UUID = Device Information>
2017-01-18 15:57:28.942 BatKeeper.iOS[2936:567501] Device.Discovered Service: <CBService: 0x17046c080, isPrimary = YES, UUID = 00001530-0000-3512-2118-0009AF100700>
2017-01-18 15:57:28.943 BatKeeper.iOS[2936:567501] Device.Discovered Service: <CBService: 0x17027ed00, isPrimary = YES, UUID = 1811>
2017-01-18 15:57:28.944 BatKeeper.iOS[2936:567501] Device.Discovered Service: <CBService: 0x17007c940, isPrimary = YES, UUID = 1802>
2017-01-18 15:57:28.944 BatKeeper.iOS[2936:567501] Device.Discovered Service: <CBService: 0x17046a880, isPrimary = YES, UUID = Heart Rate>
2017-01-18 15:57:28.945 BatKeeper.iOS[2936:567501] Device.Discovered Service: <CBService: 0x1704698c0, isPrimary = YES, UUID = FEE0>
2017-01-18 15:57:28.946 BatKeeper.iOS[2936:567501] Device.Discovered Service: <CBService: 0x17046bc40, isPrimary = YES, UUID = FEE1>
*** SearchBleServices - got services!
```
### and trace when it do not
```
2017-01-18 16:01:00.965 BatKeeper.iOS[2936:567501] Adapter: Stopping the scan for devices.
2017-01-18 16:01:00.965 BatKeeper.iOS[2936:567501] Adapter: Scan was cancelled.
** ConnectToDevice - trying to connect to MI Band 2...
2017-01-18 16:01:04.277 BatKeeper.iOS[2936:567501] ConnectedPeripherial: MI Band 2
** Device connected 1a00527a-c5b1-492f-adfd-e074e99a56b6: MI Band 2 / Connected / <CBPeripheral: 0x1740f8480, identifier = 1A00527A-C5B1-492F-ADFD-E074E99A56B6, name = MI Band 2, state = connected> :)
> Connected, searching services
*** Searching services
*** SearchBleServices - async...
2017-01-18 16:01:04.297 BatKeeper.iOS[2936:567501] ConnectToDeviceAsync Connected: 1a00527a-c5b1-492f-adfd-e074e99a56b6 MI Band 2
2017-01-18 16:01:38.282 BatKeeper.iOS[2936:567501] Disconnect error 7 Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.} CBErrorDomain
2017-01-18 16:01:38.282 BatKeeper.iOS[2936:567501] DisconnectedPeripheral by lost signal: MI Band 2
```

Here it is with a MI Band 2 device, but it is the same with other devices as Estimate iBeacon, arduino ble, Bluetooth of my Mac, ...)

## Configuration
**Version of the Plugin: 1.1.0 and 1.2.0-beta4
**Platform: iOS 10.2.1 (but also on previous version)
**Device: iPhone 7+

xabre, 2017-01-19:
Did you also notice this behavior for other devices?

BorisFR, 2017-01-19:
I have only one smartphone to play with.
I notice the same behavior with other BLE devices : iBeacon, arduino ble, ...
I have not this problem using nRF Connect or Bluefruit app.

xabre, 2017-01-24:
I will investigate your code as soon as i get some time :) Thanks for posting a link to it.

BorisFR, 2017-04-02:
Still not ok with 1.3.0-alpha1
In place of "await device.GetServicesAsync ()" I also test "await device.GetServiceAsync (Guid.Parse (SERVICE_ID))" but not better.