dotnet-bluetooth-le/dotnet-bluetooth-le #144: Refactor Android implementation and have a GattCallback for each device instead of a global one

xabre, 2017-01-19:
This should solve several issues:

- matching device id's charcateristic id's etc
- cleaner code, clearer separation, less internal events
- more async/await
- make stuff like #71 possible

alberk8, 2017-01-19:
Sign me up.  I can test for Android 5.x and 6.x.  I have a few BLE device type to test.  Thank you.

flyingxu, 2017-01-23:
This is cool, I can't wait to test.
In my app, the phone will connect to multiple devices at the same time.

allbdeveloper, 2017-01-30:
I can't wait too.

xabre, 2017-03-05:
@smstuebe Can you take a look over the changes. I did the basic refactoring involved in having a gattCallback instance / device instead of a global one.  At the moment I did not remove the existing events fired by the callback. I also managed to find an easy fix for our #147 stuck issues by using a reject handler for pending asyngs and subscribing it to the ConnectionLost event. We can discuss this and maybe also use it for iOS.

smstuebe, 2017-03-05:
I think the approach with one eventhandler is a bit too limited / special. Feels like "Ahh ok if we put this thing here it will fix it". I had more generic approach in mind that is combined with properly cleaning up everything (if this is needed).

The main question for figuring out a good solution currently is (for me): Are services / characteristics still valid after connection lost (so that I can hold the reference and just reconnect), or do they get invalid and the user has to discover everything after the reconnect?

xabre, 2017-03-05:
Services and characteristics are definitely not usable after a
disconnect/reconnect. We had this issue a while ago and as a fix we added a
clear services method on disconnect. This is valid for both iOS and android

xabre, 2017-03-15:
We released 1.3.0-alpha1 which features this refactoring. So testing can begin :) @flyingxu @alberk8 @allbdeveloper

vbisbest, 2017-05-22:
I am on 1.3.0 beta 3 and I am recieving multiple notifications for devices, services and characteristics.  Is this a regression issue or am I not understanding the original defect?  Thanks!

xabre, 2017-08-10:
@vbisbest This might be a regression. Can you describe your scenario?

closed 2019-08-05 by xabre