dotnet-bluetooth-le/dotnet-bluetooth-le #145: Proper way to use ConnectToKnownDeviceAsync for reconnection

leelatoo, 2017-01-24:
We want to reconnect to BLE devices only using Guid.

Let me exlpain the scenario which i am facing:

## Steps to reproduce
1. Turn on the BLE device
2. iOS App connects to the BLE device and stores the Guid of the BLE device in app's database
3. Kill the app.
4. Turn off BLE device.
5. Launch the app again. App tries to connect to the already known BLE device using the Guid of the ble device from database.

`CancellationTokenSource _cancellationTokenSource;
				TimeSpan time = new TimeSpan(10, 0, 0); // hours, mins, secs
				_cancellationTokenSource = new CancellationTokenSource(time);

				// Connect to device.
				IDevice device = await _adapter.ConnectToKnownDeviceAsync(guid, _cancellationTokenSource.Token);
				if (device.State == DeviceState.Connected)
				{
					Debug.WriteLine("Connected");
				}
				Debug.WriteLine("Not Connected");`

6. Turn on the Ble device.

## Expected behavior
7. Expectation is that app should get connected to the device. and DeviceConnected event of the adapter should get fired. 
According to the plugin documentation, 'ConnectToKnownDeviceAsync' API tries to connect to the Device indefinitely.

## Actual behavior
But the app does not get automatically connected to the BLE device when it is available.
 In the log, we can see attempting connection to 'guid'. But when device is truend on, connection does not happen. Please see the log below.

### log
2017-01-24 15:05:11.531 BLEAPPiOS[6500:341032]Adapter: Waiting for state: PoweredOn
2017-01-24 15:05:11.532 BLEAPPiOS[6500:341032] [Adapter] Attempting connection to ABC9D4F13-BBF7-4162-944E-C65AA6BC0027

## Configuration

**Version of the Plugin:** Plugin.BLE.1.2.0-beta3

**Platform:**  iOS 10.1.1

**Device:** i Phone 6S

Can you please help us?

xabre, 2017-01-24:
The flow you describe sound correct. One thing comes to mind: some BLE devices assign a new mac (implicitly guid) after every restart, is this the case for your device? 
This feature is being successfully used in a production app.

leelatoo, 2017-01-24:
@xabre 
Yes, it is highly probable that the BLE devices which we use, gets different address after every restart.
In this case, do you have any suggestions/ workaround to connect to those already known devices ?

xabre, 2017-01-24:
It's difficult to say, for both iOS and Android this will look like a new device and will no longer be "known".  Maybe you can talk to the guys developing the firmware of you device, they might have an idea.

leelatoo, 2017-01-25:
@xabre Thank you :)

xabre, 2017-02-01:
Did you confirm that the device gets a new ID after restart? This is indeed an interesting scenario. This might mean that the native device object keeps some information internally that allows this... unsure for now

leelatoo, 2017-03-10:
Yes, Our device gets a new address after restart. But, native device object still has the old uuid.

smstuebe, 2017-03-10:
On iOS BLE information are cached like hell. I had this issue with a device that changes the name. I had to turn Bluetooth off and on again or in some cases only a restart of the device helped. I verified this with the [light blue app ](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth-low-energy/id557428110?mt=8)

lamest, 2018-03-05:
note that to tell iOS that your device can change name or characteristics, you should implement ServiceChanged characteristic in your ble device. If it only present and does nothing, iOS device would  already clear cache on every connect.

Dinesh-Fernando, 2023-08-01:
Is there a way to get the GUID if the mac address is known?