dotnet-bluetooth-le/dotnet-bluetooth-le #146: BLE device name changed

gaminn, 2017-01-25:
Hi,
I use Android 5.1.

I discovered a device, it advertises name A.

xamarin-bluetooth-le shows this device with name A.

I changed the name of the device, it now advertises with name B.

xamarin-bluetooth-le shows this device with old name A (whereas other Android applications, e.g. nrf connect from Nordic Semiconductor, display the device with correct name B). Applications ran on the same device concurrently.

After some time (I can't specify it, maybe after restart of Android) correct name B is shown by xamarin-bluetooth-le.

It seems that xamarin-bluetooth-le reads buffered device name. How to fix it?

xabre, 2017-01-26:
The current Android implementation relies on the native BluetoothDevice.Name property, based on your issue this might be stale, or cached by Android. 
We might work around this by instead retrieving the name from the advertisement data and only fallback to the name property if there is no advertisement data.

tomh4, 2018-02-15:
Any news on this = I am experiencing the same issue, even on closing a GattServer, restarting it and having the device reconnect with its old cached name... ?

sisaacks, 2022-11-02:
any news on this......I am having the same issue

bwenninger, 2023-12-19:
Is there any news on this?

While some Android devices are working fine, a One Plus+ device running Android 12, is presenting an issue for our application.  In our situation we are connecting to a device which initially has a "hidden" name that it is advertising as.  A manual button press on the device causes it to start advertising with a new "visible" name, but the scan results in our application only give us the previous "hidden" name.

If while scanning in our application we switch to the Android Bluetooth Settings we can see the name switch from its "hidden" value to its "visible" value and then after that we can see the visible name in our application.

We do not want to force our users to switch to the Android settings page in order to be able to find the device.  Is there anyway we can clear the cache or get around it so that our app can see the changed device name?

Thanks!

Hackmodford, 2024-01-31:
I had this issue on Android and ended up writing a extension method to handle it.

```C#
public static string LatestName(this IDevice device)
{
    if (device.AdvertisementRecords != null)
    {
        foreach (var record in device.AdvertisementRecords)
        {
            if (record.Type != AdvertisementRecordType.CompleteLocalName)
            {
                continue;
            }
            if (record.Data.Length == 0)
            {
                continue;
            }
            return Encoding.UTF8.GetString(record.Data);
        }
    }
    return device.Name;
}
```