dotnet-bluetooth-le/dotnet-bluetooth-le #147: Stuck in GetServiceAsync

gaminn, 2017-01-25:
On Android 5.1 I run this procedure for my ble device:

1) ConnectToDeviceAsync
2) GetServiceAsync

During this procedure I remove power of the BLE device (so effectively the link is lost). Then xamarin-bluetooth-le gets stuck in GetServiceAsync forever.

This is very serious bug for me. Is it caused by Android or xamarin-bluetooth-le? Is there any simple fix for it? Will running xamarin-bluetooth-le methods in separate Task and cancelling the tasks after some timeout fix that?

smstuebe, 2017-01-25:
Unfortunately, there is no simple workaround. We are aware of this bug, but properly handling it, will take some time.

gaminn, 2017-01-25:
I'm sorry...

Will I face the same problem if I use native Android BLE api or is there, in Android native BLE api, the same issue?

smstuebe, 2017-01-25:
This is an issue of the plugin. We have to handle the disconnection of the device and cancel all ongoing operations that are related to this device. This handling is currently not implemented. The problem is that we convert the events into tasks. On native Android your callback doesn't simply get called, but the `GetServiceAsync` gets stuck, because it waits for this callback.

gaminn, 2017-01-25:
Thanks for the info!

xabre, 2017-01-27:
We will solve this for Android while implementing #144 :)

gaminn, 2017-01-29:
Thanks. Can you estimate release date?

smstuebe, 2017-01-29:
Not really, we are doing this in our spare time ;) We also have to discuss the correct solution and play around with it.

xabre, 2017-01-31:
I have a workaround for you...
I'll post it asap

gaminn, 2017-01-31:
Nice. Thanks!

BTW: Is there a possibility that another functions in your plugin get stuck like GetServiceAsync?

xabre, 2017-01-31:
There is a handy extension for Task that I used in some cases.
I'm writing it from memory so bare with me:
```
// returns true if the task completed before the timeout
public async Task<T> WithTimeout(this Task<T> task, int timeout)
{
     if(await Task.WhenAny(task, Task.Delay(timeout)) == task)
     {
           return task.Result;
      }
     
      throw new TimeoutException("Timed out while waiting for");      

}
```
You can use this like this:
```
try
{
   await device.GetServiceAsync(guid).WithTimeout(2000);
}
catch(TimeoutException)
{
}
catch(Exception)
{
}
```

adolgov, 2017-01-31:
I would strongly recommend looking at Polly for a robust implementation of stuff like retry/timeout/circuit-breaker/backoff/etc. In my experience, for other projects, it's been a lifesaver versus rolling self-made extensions.

https://github.com/App-vNext/Polly

benevbright, 2017-02-15:
This is a question.

I'm making an app with using XF pcl. and launched on App Store and Android Google play.

I faced sometimes a very serious issue, which app stops working "await calls inside EVERY Task.Run()" from specific time mostly after a day since the app's started. EVERY Task.Run process.

It occurs only on XF/iOS. and App is still alive. I can even move tab pages and navigation except to try Task.Run().

Unfortunately, I'm a newbie for c#. 
Any tip will help me. What can cause this issue? Have anyone experienced similar issue?

Conversations on this issue in here have any chance to occur my issue?

plus, Is this what you guys call "deadlock"?
Again, any word would help me.

I actually edit my code as xabre suggested. But still happened very rarely.

leelatoo, 2017-02-20:
I faced this problem. There is a temporary workaround which helped.

before requesting any Service, check whether the device is connected or not.

`if (DeviceState.Connected == connectedDevice.State)
			{
				IService service = await connectedDevice.GetServiceAsync(Guid.Parse(serviceUuid));
				return service;
			}`

xabre, 2019-08-05:
Reopen if still an issue after v2.0.1

closed 2019-08-05 by xabre