dotnet-bluetooth-le/dotnet-bluetooth-le #150: Device doesn't stay connected on iOS

rusty21, 2017-02-06:
I'm using this library to connect to a device but the device keeps disconnecting with this error `DisconnectedPeripheral by lost signal`. However the devices are sitting right next to each other. This happens on iOS every time. After about 2 - 10 seconds the device disconnects. On android the device stays connected so I know there is nothing wrong with the device. 
 
## Steps to reproduce

1. Connect to a device with iOS

2. Wait 2 - 10 seconds

3. Watch the device disconnect

## Expected behavior
The device should stay connected

## Actual behavior
The device drops connection even though the devices are right next to each other

## Configuration

Plugin version: 1.1.0

OS: iOS 10.2

Devices: iPhone 5, iPhone 6 Plus

Other things I've tried:
Toggling Bluetooth off and on, on the phones and the target device.
Rebooting the phones, and the target device.
Replacing batteries in target device.

I suspect you will have a hard time reproducing this. Is there anything else I can try, or something that might give you more to go on?

xabre, 2017-02-06:
@rusty21 Try to intall a 3rd party ble explorer app (like LightBlue) and try to connect your device and check if the problem still occurs. You can then post your results here :)

rusty21, 2017-02-07:
I used LightBlue like you suggested, and the device appears to stay connected. So that means either theres something wrong with my code, or the Plugin.Ble library. I've actually got a pcl that is shared between my iOS and Android app. Both platforms go through this pcl to interact with ble devices. So both platforms run the same code, but one works and the other doesn't. Any ideas?

rusty21, 2017-02-07:
RSSI seems to be between -27 and -41

rusty21, 2017-02-07:
I think I've figured it out. I'm using this code to connect to the device.

    var tokenSource = new CancellationTokenSource ();
    tokenSource.CancelAfter ((int)TimeSpan.FromSeconds (20).TotalMilliseconds);
    await CrossBluetoothLE.Current.Adapter.ConnectToDeviceAsync (device, cancellationToken: tokenSource.Token);

The token gets canceled after 20 seconds and the device disconnects. Am I doing it wrong? Is there a different way I should be canceling if a device takes longer than 20 seconds to connect?

xabre, 2017-02-07:
Ok. Nice catch :). It seems that we don't dispose the cancellation token registration even after the successfull connection, meaning that even after a successfull connect the original the token source can be used to cancel the connection.

smstuebe, 2017-02-07:
Meh. fixing this isn't done with adding a `using` :(

smstuebe, 2017-02-17:
should be fixed in 1.2.0

closed 2017-02-28 by xabre