dotnet-bluetooth-le/dotnet-bluetooth-le #151: Though BLE device sends response immediately, App receives the response after a delay.

leelatoo, 2017-02-08:
We are using plugin.ble in our app. App connects to a BLE device and upon successful connection, it reads many characteristics back to back, for example it reads 5 characetristics. 

## Steps to reproduce

The follow goes like this

send read request to characteristic1
get read response from characteristic1

send read request to characteristic2
get read response from characteristic2

send read request to characteristic3
get read response from characteristic3

send read request to characteristic4
get read response from characteristic4

send read request to characteristic5
get read response from characteristic5

## Expected behaviour
App sends read request and should get response immediately from BLE device. 

## Actual behaviour
This flow works most of time properly as expected. We verified this behaviour by using a tool which traces the BLE packets sent over the air. It works as expected in case of straight forward cases like connect to BLE device and read many characteristics.

we are dealing with a particular scenario. 

1. App is connected to BLE device, now disconnect bluetooth of iOS device. So, automatically App is disconnected from BLE device. 
2. Upon disconnection, we call ConnectToDeviceAsync(device) API to automatically get connected to BLE device when it is available.
3. Turn on iOS device's bluetooth, App gets connected to BLE device.
4. Soon after successful connection, app tries to read 5 characteristics one by one.
send read request to characteristic1
get read response from characteristic1

send read request to characteristic2
get read response from characteristic2

send read request to characteristic3
App is waiting for a response from BLE device for around 30 seconds.
After 30 seconds, app received response from characteristic3

We traced the BLE read requests and responses between app and that particular BLE device with timestamp.
**Tracing result :** App sent read request to characteristic3 at 11.48.03.108 am. BLE device sends characteristic3 read response at 11.48.03.784 am.

So, BLE device sends response to any request within a second. But, app receives the response only after around 20 to 30 seconds.

Do you have any idea about why app recieves response late? who is delaying to deliver the response to app?  Is it iOS or Plugin.BLE or App ?

This is a critical issue which we are facing now.

Can you please help us giving ideas/suggestions to solve issue?

## Configuration

**Version of the Plugin:** Plugin.BLE.1.2.0-beta3

**Device 1:**
**Platform:** e.g. iOS 9.3.1

**Device:** e.g. iPad mini 4.

**Device 2:**
**Platform:** e.g. iOS 10.1.1

**Device:** e.g. iPhone 6S.

xabre, 2017-02-10:
One idea I have at the moment is to call ConnectToKnownDeviceAsync(guid) instead of ConnectToDeviceAsync(device) this might refresh the underlying IDevice and CBPeripherial.

leelatoo, 2017-02-21:
Our BLE device will have different address after every restart of the device. So the uuid of the device is not constant.

Because of this, ConnectToKnownDeviceAsync(guid) is not an option for us.
Is there any other way to refresh underlying IDevice and CBPeripherial?

leelatoo, 2017-03-10:
Any updates on this one? We are still awaiting a reply for this and its kind of a logjam for us. Are you working on this currently? any leads would really be helpful. We checked the Release notes of Plugin.BLE 1.2.1 and we couldn't find any fix for our issue.

Appreciate your support on this!! TIA.