dotnet-bluetooth-le/dotnet-bluetooth-le #155: BLE characteristic update event fired twice

pvhd, 2017-02-17:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Connect to ble device, get specific service using GetServiceAsync(Guid.Parse("service id")), get specific characteristic using GetCharacteristicAsync(Guid.Parse("characteristic id")). Then check if characteristic can update and assign ValueUpdated event

2. Put a break point inside the ValueUpdated event and found out the event is fired twice.

## Expected behavior
The ValueUpdated event should be fired once

## Actual behavior
The ValueUpdated event is fired twice

### Crashlog

## Configuration

**Version of the Plugin:** 1.1.0 

**Platform:** Android 4.4.2

**Device:** LG G3

xabre, 2019-08-04:
Please reopen if still an issue.

closed 2019-08-04 by xabre

LinuxAndOracle, 2019-10-21:
@pvhd How did you fired twice? I can't even get in once