dotnet-bluetooth-le/dotnet-bluetooth-le #159: All of Task.run() in my project stopped working after reconnect to device on iOS rarely.

benevbright, 2017-02-28:
## Steps to reproduce

1.

On iOS, when you need to reconnect to a peripheral after you went far from the device, we do like below. And it works fine most time.

		async void OnEvent_DeviceConnectionLost(object sender, DeviceErrorEventArgs a)
		{
                        ....
				try
				{
					await Adapter.ConnectToDeviceAsync(a.Device); //for retry on iOS
				}
				catch (Exception e)
				{
					Debug.WriteLine("device exception 3 : " + e.Message);
				}
		}

## Actual behavior
Very rarely, After reconnect to a peripheral, all of Task.run() in my project stopped working on iOS.
I'm a newbie for c#.
Actually, it always happens after about a day or longer from when my app started.
It's really hard to reproduce.

I ask this question not only get an answer but also learn more about c#. 
Is this behavior the 'deadlock'?
Is this problem of plugin or Xamarin? 
Any tip will help.

Thanks.

## Configuration

**Version of the Plugin:** 1.2.0

**Platform:** e.g. iOS 10.2

**Device:** e.g. iPhone 7

xabre, 2017-02-28:
This sounds almost like the thread pool is not able to start new tasks. Might mean that you have a lot of running threads which did not terminate....

smstuebe, 2017-02-28:
What's the reason for using Task.Run? Do you **really** need a new thread for the things you are doing?

benevbright, 2017-03-01:
Oh, Thanks for your advice. 
I searched my project and I have 18 Task.Run. 
I thought I used when it's necessary. 

Anyway, Thanks. I got the answer where I start to look into.

benevbright, 2017-03-01:
@smstuebe 
I usually use Task.Run to do "await Http method" at some point. 
And the other await methods. like this.

			Task.Factory.StartNew(async () => await RequestDataFromPeripheral());

benevbright, 2017-03-01:
@xabre Hi Xabre. According to you said 'running threads are not terminated', does that occur because those threads are stopped by 'deadlock'?

smstuebe, 2017-03-01:
I think you can get rid of all these `Task.Run`. 

I assume you are doing something like:

    void OnButtonClick() 
    {
        Task.Run(async () => await httpClient.PostFooAsync("bar"));
    }

This can be simply changed to

    async void OnButtonClick() 
    {
        await httpClient.PostFooAsync("bar");
    }

The http operation is running in an IO thread and not using the UI Thread.
Please read tutorials about async/await. We can, but won't teach you here :) If you have pluralsight, watch jon skeets Talks about async/await: https://www.pluralsight.com/courses/skeet-async (Yes all of them. They will enlighten you)

benevbright, 2017-03-01:
Hi, I already watched many videos about async/await even if I'm new in c# world. 

I also have done it like your suggestion if I could and it's event handler like that example.

I think I should study more. Thanks.

benevbright, 2017-03-01:
But I asked this here, in my experience, it occurs only when I reconnect peripheral on iOS by using that way.

closed 2019-08-04 by xabre