dotnet-bluetooth-le/dotnet-bluetooth-le #160: Bad connection status

gaminn, 2017-02-28:
1) Connect to IDevice.
2) Here I perform some GATT write/read and bonding.
3) Disconnect from the IDevice.
4) Wait for disconnect event.
5) IDevice.state == Connected. The consequence is that next connect to IDevice is impossible.

xabre, 2017-02-28:
What version of the plugin are you using? Did this happen after the last update?

gaminn, 2017-02-28:
Sorry, initial post updated.

I haven't update the plugin so I can't confirm whether this issue existed in previous version.

tomridl, 2017-03-31:
@gaminn Is the problem that you can't see the device when you scan, or the next connection to the device is failing? If the device is paired to Android, then Android it self automatically created a connection to the device. Check GetSystemConnectedOrPairedDevices to see if it is already connected to the system, and if it is, try connecting to it again using ConnectToDeviceAsync.

gaminn, 2017-03-31:
I wasn't able to connect to the same IDevice object for the second time. I had to create new IDevice object (representing the same BLE device), then the connection succeeded.

tomridl, 2017-03-31:
Ok, maybe try updating the plugin. I'm using 1.2.1 on Android 5.1.1, and after calling DisconnectDeviceAsync, the number of connected devices in ConnectedDevices is 0 and the state of the IDevice changes to "Disconnected". If I then immediately try to reconnect to the same IDevice, the connection is successful. I am storing the IDevice as a static variable:

`public static IDevice ConnectedDevice { get; set; }`