dotnet-bluetooth-le/dotnet-bluetooth-le #161: Question: Issue with BLE.Client.Droid

Scuzzlebuttt, 2017-03-02:
I was just wanting to run the sample client app and test it out on my Sony Xperia z3v.  I just grabbed the latest clone from v1.2.0.

I get a compile time error in BLE.Client (Portable) in BleMvxFormsApp.xaml.cs on InitializeComponent(); in the BleMvxFormsApp() constructor:

The name 'InitializeComponent' does not exist in the current context.

Yet, I am still able to run and start the app in the Marshmallow 6.0 emulator.

But to test the bluetooth stuff I need to debug on my phone (Sony Xperia z3v).  Again, sample app complies and runs, but now throws exception.  I'm new to Xamarin.Forms, and I've never used MVVMCross before.  Looks like there is something wrong with the dependency injection, but I don't know enough to track this down.  Any help to get pushed in the right direction would be very much appreciated.  Thank you.

## Steps to reproduce

1.  Grab latest clone (v1.2.0)

2.  Run BLE.Client.Droid debugger on device.

## Expected behavior
Application should start

## Actual behavior
Application throws exception

### Crashlog

Throws MvxException in BLE.Client.Droid in MainActivity.cs on:

`Mvx.Resolve<IMvxAppStart>().Start();`

```
[0:] Failed to construct and initialize ViewModel for type BLE.Client.ViewModels.DeviceListViewModel from locator MvxDefaultViewModelLocator - check InnerException for more information
[0:] MvvmCross.Core
[0:]   at MvvmCross.Core.ViewModels.MvxViewModelLoader.LoadViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request, MvvmCross.Core.ViewModels.IMvxBundle savedState, MvvmCross.Core.ViewModels.IMvxViewModelLocator viewModelLocator) [0x00046] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxViewModelLoader.LoadViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request, MvvmCross.Core.ViewModels.IMvxBundle savedState) [0x00020] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Forms.Presenter.Core.MvxPresenterHelpers.LoadViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request) [0x00005] in D:\git\MvvmCross-Forms\MvvmCross.Forms.Presenter.Core\MvxPresenterHelpers.cs:20 
  at MvvmCross.Forms.Presenter.Core.MvxFormsPagePresenter.TryShowPage (MvvmCross.Core.ViewModels.MvxViewModelRequest request) [0x0000c] in D:\git\MvvmCross-Forms\MvvmCross.Forms.Presenter.Core\MvxFormsPagePresenter.cs:82 
  at MvvmCross.Forms.Presenter.Core.MvxFormsPagePresenter.Show (MvvmCross.Core.ViewModels.MvxViewModelRequest request) [0x00000] in D:\git\MvvmCross-Forms\MvvmCross.Forms.Presenter.Core\MvxFormsPagePresenter.cs:66 
  at MvvmCross.Droid.Views.MvxAndroidViewDispatcher+<>c__DisplayClass2_0.<ShowViewModel>b__0 () [0x00000] in <c173f0f9274a4c068a91342649d69b4c>:0 
  at MvvmCross.Droid.Views.MvxAndroidMainThreadDispatcher.RequestMainThreadAction (System.Action action) [0x00019] in <c173f0f9274a4c068a91342649d69b4c>:0 
  at MvvmCross.Droid.Views.MvxAndroidViewDispatcher.ShowViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request) [0x00014] in <c173f0f9274a4c068a91342649d69b4c>:0 
  at MvvmCross.Core.ViewModels.MvxNavigatingObject.ShowViewModelImpl (System.Type viewModelType, MvvmCross.Core.ViewModels.IMvxBundle parameterBundle, MvvmCross.Core.ViewModels.IMvxBundle presentationBundle, MvvmCross.Core.ViewModels.MvxRequestedBy requestedBy) [0x0002e] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxNavigatingObject.ShowViewModel (System.Type viewModelType, MvvmCross.Core.ViewModels.IMvxBundle parameterBundle, MvvmCross.Core.ViewModels.IMvxBundle presentationBundle, MvvmCross.Core.ViewModels.MvxRequestedBy requestedBy) [0x00000] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxNavigatingObject.ShowViewModel[TViewModel] (MvvmCross.Core.ViewModels.IMvxBundle parameterBundle, MvvmCross.Core.ViewModels.IMvxBundle presentationBundle, MvvmCross.Core.ViewModels.MvxRequestedBy requestedBy) [0x00000] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxAppStart`1[TViewModel].Start (System.Object hint) [0x00013] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at BLE.Client.Droid.MainActivity.OnCreate (Android.OS.Bundle bundle) [0x00050] in C:\Users\Brian\Documents\GitHub\xamarin-bluetooth-le\Source\BLE.Client\BLE.Client.Droid\MainActivity.cs:36 
[0:] Problem creating viewModel of type DeviceListViewModel
[0:] MvvmCross.Core
[0:]   at MvvmCross.Core.ViewModels.MvxDefaultViewModelLocator.Load (System.Type viewModelType, MvvmCross.Core.ViewModels.IMvxBundle parameterValues, MvvmCross.Core.ViewModels.IMvxBundle savedState) [0x00027] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxViewModelLoader.LoadViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request, MvvmCross.Core.ViewModels.IMvxBundle savedState, MvvmCross.Core.ViewModels.IMvxViewModelLocator viewModelLocator) [0x00015] in <c4031697dde6429f89882cbe7dd9e363>:0 
[0:] mvx:Diagnostic:  2.80 App Start
```

## Configuration

**Version of the Plugin:** 1.2.0

**Platform:** Android 5.1

**Device:** Sony Xperia z3v

smstuebe, 2017-03-02:
Ohhh nooo it's skuzzlebutt! :smile: 

Hmm can you provide the inner exception and maybe the inner exception of the inner exception and ...?

Scuzzlebuttt, 2017-03-02:
"Behold!  My Patrick Duffy leg!"

OK, so I dug deeper in the InnerExceptions:

```
[0:] Exception:
[0:] Failed to construct and initialize ViewModel for type BLE.Client.ViewModels.DeviceListViewModel from locator MvxDefaultViewModelLocator - check InnerException for more information
[0:] MvvmCross.Core
[0:]   at MvvmCross.Core.ViewModels.MvxViewModelLoader.LoadViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request, MvvmCross.Core.ViewModels.IMvxBundle savedState, MvvmCross.Core.ViewModels.IMvxViewModelLocator viewModelLocator) [0x00046] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxViewModelLoader.LoadViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request, MvvmCross.Core.ViewModels.IMvxBundle savedState) [0x00020] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Forms.Presenter.Core.MvxPresenterHelpers.LoadViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request) [0x00005] in D:\git\MvvmCross-Forms\MvvmCross.Forms.Presenter.Core\MvxPresenterHelpers.cs:20 
  at MvvmCross.Forms.Presenter.Core.MvxFormsPagePresenter.TryShowPage (MvvmCross.Core.ViewModels.MvxViewModelRequest request) [0x0000c] in D:\git\MvvmCross-Forms\MvvmCross.Forms.Presenter.Core\MvxFormsPagePresenter.cs:82 
  at MvvmCross.Forms.Presenter.Core.MvxFormsPagePresenter.Show (MvvmCross.Core.ViewModels.MvxViewModelRequest request) [0x00000] in D:\git\MvvmCross-Forms\MvvmCross.Forms.Presenter.Core\MvxFormsPagePresenter.cs:66 
  at MvvmCross.Droid.Views.MvxAndroidViewDispatcher+<>c__DisplayClass2_0.<ShowViewModel>b__0 () [0x00000] in <c173f0f9274a4c068a91342649d69b4c>:0 
  at MvvmCross.Droid.Views.MvxAndroidMainThreadDispatcher.RequestMainThreadAction (System.Action action) [0x00019] in <c173f0f9274a4c068a91342649d69b4c>:0 
  at MvvmCross.Droid.Views.MvxAndroidViewDispatcher.ShowViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request) [0x00014] in <c173f0f9274a4c068a91342649d69b4c>:0 
  at MvvmCross.Core.ViewModels.MvxNavigatingObject.ShowViewModelImpl (System.Type viewModelType, MvvmCross.Core.ViewModels.IMvxBundle parameterBundle, MvvmCross.Core.ViewModels.IMvxBundle presentationBundle, MvvmCross.Core.ViewModels.MvxRequestedBy requestedBy) [0x0002e] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxNavigatingObject.ShowViewModel (System.Type viewModelType, MvvmCross.Core.ViewModels.IMvxBundle parameterBundle, MvvmCross.Core.ViewModels.IMvxBundle presentationBundle, MvvmCross.Core.ViewModels.MvxRequestedBy requestedBy) [0x00000] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxNavigatingObject.ShowViewModel[TViewModel] (MvvmCross.Core.ViewModels.IMvxBundle parameterBundle, MvvmCross.Core.ViewModels.IMvxBundle presentationBundle, MvvmCross.Core.ViewModels.MvxRequestedBy requestedBy) [0x00000] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxAppStart`1[TViewModel].Start (System.Object hint) [0x00013] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at BLE.Client.Droid.MainActivity.OnCreate (Android.OS.Bundle bundle) [0x00050] in C:\Users\Brian\Documents\GitHub\xamarin-bluetooth-le\Source\BLE.Client\BLE.Client.Droid\MainActivity.cs:36 
[0:] Exception.InnerException:
[0:] Problem creating viewModel of type DeviceListViewModel
[0:] MvvmCross.Core
[0:]   at MvvmCross.Core.ViewModels.MvxDefaultViewModelLocator.Load (System.Type viewModelType, MvvmCross.Core.ViewModels.IMvxBundle parameterValues, MvvmCross.Core.ViewModels.IMvxBundle savedState) [0x00027] in <c4031697dde6429f89882cbe7dd9e363>:0 
  at MvvmCross.Core.ViewModels.MvxViewModelLoader.LoadViewModel (MvvmCross.Core.ViewModels.MvxViewModelRequest request, MvvmCross.Core.ViewModels.IMvxBundle savedState, MvvmCross.Core.ViewModels.IMvxViewModelLocator viewModelLocator) [0x00015] in <c4031697dde6429f89882cbe7dd9e363>:0 
[0:] Exception.InnerException.InnerException:
[0:] Need BLUETOOTH permission: Neither user 10402 nor current process has android.permission.BLUETOOTH.
[0:] mscorlib
[0:]   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/3511/501e63ce/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
  at Java.Interop.JniEnvironment+InstanceMethods.CallIntMethod (Java.Interop.JniObjectReference instance, Java.Interop.JniMethodInfo method, Java.Interop.JniArgumentValue* args) [0x00085] in /Users/builder/data/lanes/3511/501e63ce/source/Java.Interop/src/Java.Interop/Java.Interop/JniEnvironment.g.cs:11484 
  at Java.Interop.JniPeerMembers+JniInstanceMethods.InvokeAbstractInt32Method (System.String encodedMember, Java.Interop.IJavaPeerable self, Java.Interop.JniArgumentValue* parameters) [0x00017] in /Users/builder/data/lanes/3511/501e63ce/source/Java.Interop/src/Java.Interop/Java.Interop/JniPeerMembers.JniInstanceMethods_Invoke.cs:195 
  at Android.Bluetooth.BluetoothAdapter.get_State () [0x00000] in /Users/builder/data/lanes/3511/501e63ce/source/monodroid/src/Mono.Android/platforms/android-23/src/generated/Android.Bluetooth.BluetoothAdapter.cs:370 
  at Plugin.BLE.BleImplementation.GetInitialStateNative () [0x00012] in C:\Users\Brian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Android\BleImplementation.cs:35 
  at Plugin.BLE.Abstractions.BleImplementationBase.Initialize () [0x00008] in C:\Users\Brian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\BleImplementationBase.cs:41 
[... cut ...]
	at android.app.ActivityThread.main(ActivityThread.java:5373)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1020)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:815)
```
And now it becomes pretty obvious what the problem is (duh).  I'll see if I can't figure out how to give the necessary permissions (thought that was only for Marshmallow and newer).

Scuzzlebuttt, 2017-03-02:
I went in to the BLE.Client.Droid project properties and checked off the "BLUETOOTH" required permission in the Android Manifest.

This allowed the app to start up without exception.  However, its not seeing my connected bluetooth device (a Zodiac Selfie Remote Shutter clicker, which I assume is LE).

Got this in the output window if it is telling:

```
[0:] mvx:Diagnostic:  4.24 Caution: GetSystemConnectedDevices does not take into account the 'services' parameter on Android.
03-02 09:35:18.746 D/BluetoothManager(24523): getConnectedDevices
[0:] mvx:Diagnostic:  4.26 Caution: GetSystemConnectedDevices does not take into account the 'services' parameter on Android.
03-02 09:35:18.762 D/BluetoothManager(24523): getConnectedDevices
[0:] mvx:Diagnostic:  4.27 Caution: GetSystemConnectedDevices does not take into account the 'services' parameter on Android.
03-02 09:35:18.776 D/BluetoothManager(24523): getConnectedDevices
```

I really appreciate your help!!!  Thank you!

Scuzzlebuttt, 2017-03-02:
Looks like the Zodiac is connecting and emulating a keyboard.  Because of that, can I not access it via BLE?

Again, thank you for your time and I apologize for asking noob questions...

closed 2024-04-09 by Scuzzlebuttt