dotnet-bluetooth-le/dotnet-bluetooth-le #162: Question : Avoid App Crash

ClausElmann, 2017-03-02:
Hi

Thanks for all your effort in creating this great plugin.
I would like to avoid to crash the app, by catching errors.
It is typically when disconnected from one device and then connected to another
It this not possible to avoid a total crash?

Thanks 

java.lang.reflect.InvocationTargetException
--- End of managed Java.Lang.RuntimeException stack trace --- java.lang.RuntimeException: java.lang.reflect.InvocationTargetException at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:689) Caused by: java.lang.reflect.InvocationTargetException at java.lang.reflect.Method.invoke(Native Method) at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:799) ... 1 more Caused by: android.runtime.JavaProxyThrowable: Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED. at Plugin.BLE.Android.Characteristic.InternalWrite (System.Byte[] data) [0x00054] in <3aa2b548c5df42f7b6b7525cb875a077>:0 at Plugin.BLE.Android.Characteristic+<>c__DisplayClass19_0.b__0 () [0x00000] in <3aa2b548c5df42f7b6b7525cb875a077>:0 at Plugin.BLE.Abstractions.Utils.TaskBuilder+d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext () [0x000b3] in <64005c76ba5d40a0a120267b87cbeb04>:0 --- End of stack trace from previous location where exception was thrown --- at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at Plugin.BLE.Android.Characteristic+d__19.MoveNext () [0x000dc] in <3aa2b548c5df42f7b6b7525cb875a077>:0 --- End of stack trace from previous location where exception was thrown --- at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at Plugin.BLE.Abstractions.CharacteristicBase+d__31.MoveNext () [0x000b1] in <64005c76ba5d40a0a120267b87cbeb04>:0 --- End of stack trace from previous location where exception was thrown --- at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0004e] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at Mobile.Portable.BleClient+<>c__DisplayClass18_0+<b__0>d.MoveNext () [0x00075] in :0 --- End of stack trace from previous location where exception was thrown --- at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at System.Runtime.CompilerServices.AsyncMethodBuilderCore.m__0 (System.Object state) [0x00000] in <4c2dbb4827d24eee9622a2b2ee97c659>:0 at Android.App.SyncContext+c__AnonStorey0.<>m__0 () [0x00000] in :0 at Java.Lang.Thread+RunnableImplementor.Run () [0x0000b] in :0 at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00009] in :0 at (wrapper dynamic-method) System.Object:d5da05c1-f779-491e-9bc7-bdff5f79470d (intptr,intptr) at mono.java.lang.RunnableImplementor.n_run(Native Method) at mono.java.lang.RunnableImplementor.run(RunnableImplementor.java:30) at android.os.Handler.handleCallback(Handler.java:743) at android.os.Handler.dispatchMessage(Handler.java:95) at android.os.Looper.loop(Looper.java:150) at android.app.ActivityThread.main(ActivityThread.java:5639) ... 3 more 

**Version of the Plugin:**  1.0.2 beta4

xabre, 2017-03-03:
Most Ble API calls might throw an exception if something goes wrong. This is how we built the API. It would be good practice to surround calls in try catch blocks and to some exception handling. 
You might also want to update the plugin to the latest version.

xabre, 2017-04-02:
Also take a look at #180. There is some info regarding the `Gatt write characteristic FAILED` exception.

closed 2017-04-02 by xabre