dotnet-bluetooth-le/dotnet-bluetooth-le #164: StartScanningForDevicesAsync not working on Android 4.3

alextn, 2017-03-09:
I use StartScanningForDevicesAsync() in my project and it works on my tablet Asus Nexus 7 Android 6.0.1 & Samsung Galaxy Tab A android 6.0.1 ; but it doesn't work on my phone Alcatel Onetouch  5050X android 4.3.

## Steps to reproduce

1. use this code:
```
// Scan devices
_cancellationTokenSource = new CancellationTokenSource();
MainController.BleController.Adapter.ScanTimeout = 5000; // scan timer 5 sec
Guid[] guidArray = new Guid[] { new Guid(MY_DEVICE_SERVICE_GUID) }; // Filter on service GUID
await MainController.BleController.Adapter.StartScanningForDevicesAsync(guidArray, null, false, _cancellationTokenSource.Token); // [breakpoint here ; start of log farther]

if (DiscoveredDevices.Count >= 1) // [breakpoint here ; end of log farther]
{...}
```

2. with this AndroidManifest.xml
```
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android" android:versionCode="X" android:versionName="Y" package="MY_PACKAGE">
	<uses-sdk android:minSdkVersion="18" android:targetSdkVersion="23" />
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
	<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
	<application android:label="MY_APPLI" android:icon="@drawable/icon" android:theme="@style/MY_THEME"></application>
</manifest>
```

## Expected behavior
Find my device(s) & DiscoveredDevices.Count >= 1

## Actual behavior
Find my device(s) with android 6 (DiscoveredDevices.Count >=1)

but not with android 4.3 (DiscoveredDevices.Count = 0).

### Crashlog
Android 6.0.1:
> mvx:Diagnostic: 19,12 Adapter >=21: Starting a scan for devices with service Id MY_DEVICE_SERVICE_GUID.
> [BluetoothAdapter] STATE_ON
> [BluetoothLeScanner] onClientRegistered() - status=0 clientIf=5
> [Choreographer] Skipped 562 frames!  The application may be doing too much work on its main thread.
> [Mono] DllImport searching in: '__Internal' ('(null)').
> [Mono] Searching for 'java_interop_jnienv_is_assignable_from'.
> [Mono] Probing 'java_interop_jnienv_is_assignable_from'.
> [Mono] Found as 'java_interop_jnienv_is_assignable_from'.
> [Mono] DllImport searching in: '__Internal' ('(null)').
> [Mono] Searching for 'java_interop_jnienv_get_byte_array_region'.
> [Mono] Probing 'java_interop_jnienv_get_byte_array_region'.
> [Mono] Found as 'java_interop_jnienv_get_byte_array_region'.
> [Mono] Unloading image System.Runtime.Extensions.dll [0x990fba00].
> [Mono] Image addref System.Runtime.Extensions[0x93adcfe0] -> System.Runtime.Extensions.dll[0x9c710700]: 8
> [Mono] Config attempting to parse: 'System.Runtime.Extensions.dll.config'.
> [Mono] Config attempting to parse: '/Users/builder/data/lanes/4009/f3074d2c/source/monodroid/builds/install/mono-armv7/etc/mono/assemblies/System.Runtime.Extensions/System.Runtime.Extensions.config'.
> [Mono] Assembly Ref addref Plugin.BLE.Abstractions[0xaa616e60] -> System.Runtime.Extensions[0x9c733780]: 6
> mvx:Diagnostic: 20,11 Adv rec [Type Flags; Data 06]
> mvx:Diagnostic: 20,12 Adv rec [Type UuidsComplete128Bit; Data 00-11-0E-80-57-31-11-E5-88-9E-00-02-A5-ZZ-YY-XX]
> ===================== DeviceListViewModel.OnDeviceDiscovered:  - 00000000-0000-0000-0000-001ecXXXXXX1
> [Mono] Unloading image System.Linq.dll [0x990fc900].
> [Mono] Image addref System.Linq[0x93add1c0] -> System.Linq.dll[0x9cc15500]: 12
> [Mono] Config attempting to parse: 'System.Linq.dll.config'.
> [Mono] Config attempting to parse: '/Users/builder/data/lanes/4009/f3074d2c/source/monodroid/builds/install/mono-armv7/etc/mono/assemblies/System.Linq/System.Linq.config'.
> [Mono] Assembly Ref addref Library_BLE_Xamarin[0xaa6170a0] -> System.Linq[0x9ca958a0]: 10
> mvx:Diagnostic: 20,30 Adv rec [Type Flags; Data 06]
> mvx:Diagnostic: 20,30 Adv rec [Type UuidsComplete128Bit; Data 00-11-0E-80-57-31-11-E5-88-9E-00-02-A5-ZZ-YY-XX]
> ===================== DeviceListViewModel.OnDeviceDiscovered:  - 00000000-0000-0000-0000-001ecYYYYYY2
> mvx:Diagnostic: 20,34 Adv rec [Type Flags; Data 06]
> mvx:Diagnostic: 20,35 Adv rec [Type UuidsComplete128Bit; Data 00-11-0E-80-57-31-11-E5-88-9E-00-02-A5-ZZ-YY-XX]
> ===================== DeviceListViewModel.OnDeviceDiscovered:  - 00000000-0000-0000-0000-001ecZZZZZZ3
> [Mono] [0x9649c930] hill climbing, change max number of threads 5
> mvx:Diagnostic: 24,23 Adapter: Scan timeout has elapsed.
> mvx:Diagnostic: 24,23 Adapter: Stopping the scan for devices.
> mvx:Diagnostic: 24,24 Adapter >= 21: Stopping the scan for devices.
> [BluetoothAdapter] STATE_ON

Android 4.3:
> [Mono] [0x6405c450] worker starting
> mvx:Diagnostic:3205,37 Adapter < 21: Starting a scan for devices.
> [BluetoothAdapter] startLeScan(): [Ljava.util.UUID;@418512e0
> [BluetoothAdapter] onClientRegistered() - status=0 clientIf=4
> [Choreographer] Skipped 300 frames!  The application may be doing too much work on its main thread.
> mvx:Diagnostic:3210,49 Adapter: Scan timeout has elapsed.
> mvx:Diagnostic:3210,50 Adapter: Stopping the scan for devices.
> mvx:Diagnostic:3210,50 Adapter < 21: Stopping the scan for devices.
> [BluetoothAdapter] stopLeScan()

## Configuration

**Version of the Plugin:** Plugin.BLE version 1.2.0

**Platform:** Android 4.3 ; Android 6.0.1

**Device:** Alcatel Onetouch 5050X ; Asus Nexus 7 & Samsung Galaxy Tab A

xabre, 2017-03-15:
There is a known android 4.3 ble bug where setting the serices filter when scanning does not yield any results. You might need to work around this issue for Android 4.3 by not using the services parameter and instead filterin by using the advertisment data.

xabre, 2017-03-30:
Any luck?

alextn, 2017-03-31:
Thanks for your reply ; no more luck here but I use a workaround:

```
if (Device.OS == TargetPlatform.Android)
{
	if (CrossDeviceInfo.Current.VersionNumber.Major <= 4 && CrossDeviceInfo.Current.VersionNumber.Minor < 4)
	{
		// BLE bug with android 4.3 so use Scan without filter
		_cancellationTokenSource = new CancellationTokenSource();
		await MainController.BleController.Adapter.StartScanningForDevicesAsync(_cancellationTokenSource.Token);
	}
}
```

xabre, 2017-03-31:
@alextn I will be closing this issue and add the findings on our limitations/remarks section in the readme.

closed 2017-03-31 by xabre