dotnet-bluetooth-le/dotnet-bluetooth-le #165: Problem with many service

Nuada26, 2017-03-10:
Hi all,

 I use bluetooth Le for get data from an object.

 I have 5 service and i have characteristic on all service, on 3 service i have characteristic with notification, but when i "StartUpdatesAsync" on 2 service my application crash, it's like i can be notify by 2 service on same time .

This is my initialisation of my services :

`svc1= new BleSvc_Sensors(await e.Device.GetServiceAsync(Guid.Parse(BleCustomUUID.SVC_SENSOR)));
            svc1.OnSvcValueUpdate += svc1_OnSvcValueUpdate;
            svc2= new BleSvc_Battery(await e.Device.GetServiceAsync(Guid.Parse("0000180f-0000-1000-8000-00805f9b34fb")));
            svc2.OnSvcValueUpdate += svc2_OnSvcValueUpdate;
            svc3= new BleSvc_DataTransfert(await e.Device.GetServiceAsync(Guid.Parse(BleCustomUUID.SVC_DATA)));
            svc4= new BleSvc_Thresholds(await e.Device.GetServiceAsync(Guid.Parse(BleCustomUUID.SVC_THRESHOLDS)));
            svc5= new BleSvc_Alarm(await e.Device.GetServiceAsync(Guid.Parse(BleCustomUUID.SVC_ALARM)));
             svc5.OnSvcValueUpdate +=svc5_OnSvcValueUpdate;
            svc6= new BleSvc_UpdateFrequencies(await e.Device.GetServiceAsync(Guid.Parse(BleCustomUUID.SVC_UPDATE)));`

and i give on exemple of one services 

`class BleSvc_Sensors
    {
        private IService _service;
        private ICharacteristic accelearationsensorcharach;
        private ICharacteristic gpssensorcharach;
        private ICharacteristic gyroscopesensorcharach;
        private ICharacteristic humiditysensorcharach;
        private ICharacteristic tempsensorcharach;

        public event OnSvcValueUpdatedHandler OnSvcValueUpdate;

        public BleSvc_MuSensors(IService service)
        {
            _service = service;
            Initialize();
        }

        public async void Initialize()
        {

            accelearationsensorcharach = await _service.GetCharacteristicAsync(Guid.Parse(BleCustomUUID.CHAR_SENSOR_ACCELARATION));
            gpssensorcharach = await _service.GetCharacteristicAsync(Guid.Parse(BleCustomUUID.CHAR__SENSOR_GPS));
            gyroscopesensorcharach = await _service.GetCharacteristicAsync(Guid.Parse(BleCustomUUID.CHAR_SENSOR_GYROSCOPE));
            humiditysensorcharach = await _service.GetCharacteristicAsync(Guid.Parse(BleCustomUUID.CHAR_SENSOR_HUMIDITY));
            tempsensorcharach = await _service.GetCharacteristicAsync(Guid.Parse(BleCustomUUID.CHAR_SENSOR_TEMP));

            accelearationsensorcharach.ValueUpdated += Accelearationsensorcharach_ValueUpdated;
            gpssensorcharach.ValueUpdated += Gpssensorcharach_ValueUpdated;
            gyroscopesensorcharach.ValueUpdated += Gyroscopesensorcharach_ValueUpdated;
            humiditysensorcharach.ValueUpdated += Humiditysensorcharach_ValueUpdated;
            tempsensorcharach.ValueUpdated += Tempsensorcharach_ValueUpdated;
            
            await accelearationsensorcharach.StartUpdatesAsync();
            await gpssensorcharach.StartUpdatesAsync();
            await gyroscopesensorcharach.StartUpdatesAsync();
            await tempsensorcharach.StartUpdatesAsync();
            await humiditysensorcharach.StartUpdatesAsync();
        }

        private void Tempsensorcharach_ValueUpdated(object sender, Plugin.BLE.Abstractions.EventArgs.CharacteristicUpdatedEventArgs e)
        {
            OnSvcValueUpdate(this, new SensorValueChangedArgs(SensorValueChangedArgs.ValueChangedList.TEMPERATURE));
        }

        private void Humiditysensorcharach_ValueUpdated(object sender, Plugin.BLE.Abstractions.EventArgs.CharacteristicUpdatedEventArgs e)
        {
            OnSvcValueUpdate(this, new SensorValueChangedArgs(SensorValueChangedArgs.ValueChangedList.HUMIDITY));
        }

        private void Gyroscopesensorcharach_ValueUpdated(object sender, Plugin.BLE.Abstractions.EventArgs.CharacteristicUpdatedEventArgs e)
        {
            OnSvcValueUpdate(this, new SensorValueChangedArgs(SensorValueChangedArgs.ValueChangedList.GYROSCOPE));
        }

        private void Gpssensorcharach_ValueUpdated(object sender, Plugin.BLE.Abstractions.EventArgs.CharacteristicUpdatedEventArgs e)
        {SensorValueChangedArgs(SensorValueChangedArgs.ValueChangedList.GPS));
        }

        private void Accelearationsensorcharach_ValueUpdated(object sender, Plugin.BLE.Abstractions.EventArgs.CharacteristicUpdatedEventArgs e)
        {
            OnSvcValueUpdate(this, new SensorValueChangedArgs(SensorValueChangedArgs.ValueChangedList.ACCELEROMETER));
        }

        
    }`

I use Wiko Fever for test this application. I have this on exit :

`03-10 11:42:38.877 D/BluetoothGatt( 8694): onSearchComplete() = Device=02:80:E1:00:34:13 Status=0
03-10 11:42:38.877 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.891 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.891 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.891 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.891 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.891 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.891 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.891 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.891 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:38.977 D/BluetoothGatt( 8694): setCharacteristicNotification() - uuid: d6898de3-6e8a-11e6-bdf4-0800200c9a66 enable: true
03-10 11:42:38.977 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
Resolved pending breakpoint at 'BleSvc_Battery.cs:31,1' to void XamarinBleLibrary.BleSvc_Battery.Batterylvlcharach_ValueUpdated (object sender, Plugin.BLE.Abstractions.EventArgs.CharacteristicUpdatedEventArgs e) [0x00000].
03-10 11:42:39.064 D/BluetoothGatt( 8694): setCharacteristicNotification() - uuid: 00002a19-0000-1000-8000-00805f9b34fb enable: true
03-10 11:42:39.064 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.080 D/BluetoothGatt( 8694): setCharacteristicNotification() - uuid: e34a3df2-6f4c-11e6-bdf4-0800200c9a66 enable: true
03-10 11:42:39.080 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.110 D/BluetoothGatt( 8694): setCharacteristicNotification() - uuid: 8d17b731-6e8b-11e6-bdf4-0800200c9a66 enable: true
03-10 11:42:39.110 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.138 D/Mono    ( 8694): DllImport searching in: '__Internal' ('(null)').
03-10 11:42:39.138 D/Mono    ( 8694): Searching for 'java_interop_jnienv_new_byte_array'.
03-10 11:42:39.138 D/Mono    ( 8694): Probing 'java_interop_jnienv_new_byte_array'.
03-10 11:42:39.138 D/Mono    ( 8694): Found as 'java_interop_jnienv_new_byte_array'.
03-10 11:42:39.141 D/BluetoothGatt( 8694): writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
03-10 11:42:39.141 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.168 D/BluetoothGatt( 8694): writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
03-10 11:42:39.168 D/BluetoothGatt( 8694): writeDescriptor: mDeviceBusy = true, and return false
03-10 11:42:39.184 D/BluetoothGatt( 8694): writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
03-10 11:42:39.184 D/BluetoothGatt( 8694): writeDescriptor: mDeviceBusy = true, and return false
03-10 11:42:39.233 D/BluetoothGatt( 8694): onDescriptorWrite() - Device=02:80:E1:00:34:13 UUID=d6898de3-6e8a-11e6-bdf4-0800200c9a66 status=0 srvcType=0 srvcInstId=0 charInstId=0 descrInstId=0
03-10 11:42:39.233 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.233 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.233 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.233 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.234 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13
03-10 11:42:39.234 D/BluetoothDevice( 8694): mAddress: 02:80:E1:00:34:13`

xabre, 2017-03-15:
The BLE stack implementation on both Android and  in general have a limit on home many characteristics can notify at the same time, I'm not sure specifically what that number is but my guess is that it's ~4 (for Android 4.x). Maybe you can try just subscribing to 4 andd see if it works. What version of Android does your phone run?

If this is the case, you want to get notifications from 5 characterisitcs maybe you should redesing your device and have a dedicated notification characterisitc which tells you the sensor type that changed and the new value ... @smstuebe what do you think?

Nuada26, 2017-03-15:
Hi i'm on 6.0 version of android. When i try with on service it's works . But before i use a library made by myself and i can be notify of all service in my case. But i can't with your package.

xabre, 2017-03-15:
Hmm, based on your log outputs it seems that the StartUpdatesAsync are not waiting for eachother to complete.... hence `writeDescriptor: mDeviceBusy = true` . Can you add some await Task.Delay(>250) between the await StartUpdatesAsync calls, just so that we can confirm this?

Nuada26, 2017-03-28:
Hi, 

Sorry for be late, but u put me on the right way.

I explain:

After i put Task.Delay(2000), i see nothing change. 
But we can't know the time for write in descriptor.

So i changed my function initilaze() from void to Task.
and when i get service of device i create object and after i await initiliaze function.
So all StartUpdatesAsync calls wait the end of other. And know it's works !

Thank you so much for put me in the right way !