dotnet-bluetooth-le/dotnet-bluetooth-le #168: system.TypeLoadException

simson51, 2017-03-14:
After updating to 1.2.1 we get a "system.TypeLoadException": Could not resolve type with token 01000061

` at System.Runtime.CompilerServices.AsyncTaskMethodBuilder.Start[TStateMachine] (TStateMachine& stateMachine) [0x00031] in /Users/builder/data/lanes/4009/f3074d2c/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/AsyncMethodBuilder.cs:316 
  at somebiz.ble_device.ble.BluetoothConnection.dosomething (somebiz.ble_device.bo.Key key) [0x00017] in <c41fd53a4c3b4c49aeddc457e2a13ea1>:0 
  at somebiz.ble_device.ble.BluetoothConnection+<Open>c__async0.MoveNext () [0x00022] in /bitbucket/somebiz/somebiz.ble_device/BLE/BluetoothConnection.cs:44 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/4009/f3074d2c/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
  at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<ThrowAsync>m__0 (System.Object state) [0x00000] in /Users/builder/data/lanes/4009/f3074d2c/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/AsyncMethodBuilder.cs:1018 
  at Android.App.SyncContext+<Post>c__AnonStorey0.<>m__0 () [0x00000] in /Users/builder/data/lanes/4009/9578cdcd/source/xamarin-android/src/Mono.Android/Android.App/SyncContext.cs:35 
  at Java.Lang.Thread+RunnableImplementor.Run () [0x0000b] in /Users/builder/data/lanes/4009/9578cdcd/source/xamarin-android/src/Mono.Android/Java.Lang/Thread.cs:36 
  at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00009] in /Users/builder/data/lanes/4009/9578cdcd/source/monodroid/src/Mono.Android/platforms/android-25/src/generated/Java.Lang.IRunnable.cs:81 
  at (wrapper dynamic-method) System.Object:f8f4278c-544d-460d-aa11-272e1a2fad38 (intptr,intptr)`

xabre, 2017-03-15:
@simson51 Can you maybe paste a code sample? thx.

simson51, 2017-03-15:
Hmm, it's not that easy (company policy). I've got a class with an IAdapter as a global var. The class itself is a singleton and contains, among other, an async Method. As soon as I call it I get the exception. I've tried to figure out what's going on there, but due to high workload I could only revert to 1.2.0.
We've got two classes working with ble. One, the scanner, works fine. The second one, where we want to connect to the devices does not. I'm running .NET Portable PCL 4.6 - Profile151.

xabre, 2017-03-31:
@simson51 
We bumped our target SDK to 7.1 so maybe this might cause your issue.

closed 2019-08-07 by xabre