dotnet-bluetooth-le/dotnet-bluetooth-le #169: ArgumentNullException when Characteristic doesn't return any value

Cheesebaron, 2017-03-14:
I have a silly device which in some cases does not return anything from specific characteristics. No error or ok from it. This makes the plugin blow up the application when that happens with a ArgumentNullException.

Stack Trace:

```
Application Specific Information:
*** Terminating app due to uncaught exception 'System.ArgumentNullException', reason: 'System.ArgumentNullException: Value cannot be null.'

Xamarin Exception Stack:
Parameter name: src
  at (wrapper managed-to-native) System.Runtime.InteropServices.Marshal:copy_from_unmanaged (intptr,int,System.Array,int)
  at System.Runtime.InteropServices.Marshal.Copy (System.IntPtr source, System.Byte[] destination, System.Int32 startIndex, System.Int32 length) [0x00000] in /Library/Frameworks/Xamarin.iOS.framework/Versions/10.4.0.123/src/mono/mcs/class/corlib/System.Runtime.InteropServices/Marshal.cs:159 
  at Foundation.NSData.ToArray () [0x00012] in /Users/builder/data/lanes/3985/35d1ccd0/source/xamarin-macios/src/Foundation/NSData.cs:50 
  at Plugin.BLE.iOS.Characteristic.get_Value () [0x00000] in C:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.iOS\Characteristic.cs:25 
  at Plugin.BLE.iOS.Characteristic+<>c__DisplayClass15_0.<ReadNativeAsync>b__2 (System.Object sender, CoreBluetooth.CBCharacteristicEventArgs args) [0x00051] in C:\Projects\xamarin-bluetooth-le\Source\Plugin.BLE.iOS\Characteristic.cs:71 
  at CoreBluetooth.CBPeripheral+_CBPeripheralDelegate.UpdatedCharacterteristicValue (CoreBluetooth.CBPeripheral peripheral, CoreBluetooth.CBCharacteristic characteristic, Foundation.NSError error) [0x00015] in /Users/builder/data/lanes/3985/35d1ccd0/source/xamarin-macios/src/build/ios/native/CoreBluetooth/CBPeripheral.g.cs:497 
  at (wrapper managed-to-native) UIKit.UIApplication:UIApplicationMain (int,string[],intptr,intptr)
  at UIKit.UIApplication.Main (System.String[] args, System.IntPtr principal, System.IntPtr delegate) [0x00005] in /Users/builder/data/lanes/3985/35d1ccd0/source/xamarin-macios/src/UIKit/UIApplication.cs:79 
  at UIKit.UIApplication.Main (System.String[] args, System.String principalClassName, System.String delegateClassName) [0x00038] in /Users/builder/data/lanes/3985/35d1ccd0/source/xamarin-macios/src/UIKit/UIApplication.cs:63 
  at SetupCompanion.Touch.Application.Main (System.String[] args) [0x00001] in C:\vcs\git\NMT Setup\src\SetupCompanion.Touch\Main.cs:19
```

## Expected behavior
I would expect the app to happily continue functioning or the Characteristic failing because nothing was read. Instead of the exception being thrown somewhere I can't catch it.

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 1.1.0

**Platform:** iOS 10.2

**Device:** iPhone 6

smstuebe, 2017-03-15:
Just to confirm. You do something like:

	try {
		// ...
		var value = await characteristic.ReadAsync();
	}
	catch(Exception e){
		// not catched?!
	}

and the App crashes?

Cheesebaron, 2017-03-15:
I do exactly that and it still crashes, yes.

Cheesebaron, 2017-03-16:
Uhm, I think the debugger is confused about this one. The exception was actually thrown somewhere else, by a command that was null... Strange that the stack trace was derpy like that...

closed 2017-03-16 by Cheesebaron