dotnet-bluetooth-le/dotnet-bluetooth-le #173: ValueUpdated event not fired on IOS for a notify characteristic

one-id, 2017-03-17:
I'm trying to read the value of a notify characteristic. I'm using the ValueUpdated event handler, everything works fine on android. On IOS, the event handler is never called. 

It also works correctly with a native code and the same devices (IOS device & BLE peripheral).

Am I missing something in the BLE plugin ? Anyone experienced the same problem ?

## Steps to reproduce

1. Register a ValueUpdated event handler on a notify characteristic. 
ICharacteristic lectureChar = await BTmodMainService.GetCharacteristicAsync(Guid.Parse(readCharGuid));

lectureChar.ValueUpdated += (o, args) =>
                        {
                             [...]
                        };

2. Start listening
 await lectureChar.StartUpdatesAsync();

3. The event Handler is never called.

## Expected behavior

The ValueUpdated event should be fired when the notify characteristic is updated.

## Actual behavior

Nothing happens.

### Crashlog

Empty.

## Configuration

**Version of the Plugin: 1.2.1 (Vanilla)

**Platform: IOS 10

**Device: IPad Air 2, Iphone 6

xabre, 2017-03-17:
@one-id Did you try to use our sample app to see if the updates are working? I have just tried the sample app with our test BLE device and everything seems to be fine.

one-id, 2017-03-20:
Ok, it's working with the sample app, thanks for your answer. I checked my app and I'm also receiving the updates now, but not when I'm expecting them. 

Here is my scenario: 
    1. I'm sending a request to the BLE device (with a writeAsync)
    2. I'm listening to the notify characteristic, storing all byte received in a buffer and waiting for a "stop" character
   3. Return the buffer

On Android, everythings work fine. The updates are fired immediatly and my "endChar" variable is updated.

On IOS, I reached the timeout (I tried to increase it) without any updates. All the update events are fired after the timeout and the completion of my sendRequestWithresponse function.   

I think I'm not doing something correctly. How can I handle his kind of scenario ?

```
public async Task<String> sendRequestWithResponse(string request)
        {
                     endChar= false;
                     receptionBuffer = "";

                    // send the request to the BLE device (WriteAsync)
                    await sendRequest(request);                    

                    sw = new Stopwatch();
                    sw.Start();

                    // Waiting for the endChar to be received or the timeout
                    while (!endChar)
                    {
                          
                           // the ValueUpdated should be fired here, and update the enChar and receptionBuffer variables     
                         
                        if (sw.ElapsedMilliseconds > readTimeOut)
                        {
                            endChar= true;
                        }
                    }
                    
                    sw.Stop();
                return receptionBuffer;           
        }

                 String res = "";
                Task<String> lectureInst = manager.envoiTrameAvecAttenteReponse(TexteToSend.Text + "\r");
                
                res = await lectureInst;
```

xabre, 2017-03-27:
@one-id 
The while loop in your code is quite dangerous and might block. I would use a TaskCompletionSource, await it in you code and set the result in the CharacteristicValueUpdated event.

one-id, 2017-03-29:
Thanks a lot, it's working with the TaskCompletionSource. 

Here is my updated code: 

```
public async Task<String> sendRequestWithResponse(string request)
        {
                     endChar= false;
                     receptionBuffer = "";
                     trameCompletion = new TaskCompletionSource<String>();

                    // send the request to the BLE device (WriteAsync)
                    await sendRequest(request);                    
                   
                 return await trameCompletion.Task;           
        }
```

And inside my CharacteristicOnValueUpdated function, I'm calling :

`trameCompletion.SetResult(receptionBuffer);
`

closed 2017-03-29 by one-id

Duality4Y, 2017-06-29:
@one-id 
you said:
> Ok, it's working with the sample app, thanks for your answer. I checked my app and I'm also receiving the updates now, but not when I'm expecting them.

what did you do to get it working ?

rakesh5689, 2019-10-02:
So, I have the same problem, in iOS at this line : await characteristic.WriteAsync(data, cancellationToken); it just seems to go nowhere from here
In Android it works fine