dotnet-bluetooth-le/dotnet-bluetooth-le #176: StartUpdatesAsync on characteristic Android error

joriano, 2017-03-21:
## Steps to reproduce

1. Download  xamarin-bluetooth-le from master branch

2. Launch BLE explorer 

3. Coonect to a Device with custom service - with 2 characteristics
- 1 Write without Response (1)
- 1 with Notify (2)

4. try to write and update

## Expected behavior

Write on characteristic 1 send data to my device
Update on characteristic 2 get data when my device send data

## Actual behavior

I connected to my device, discover services, and get characteristics.
- carac 1 : Write without Response (1)
- carac 2 : with Notify (2)

when i wrote on mu (1), its ok
when i try to update (2), my application give an exeption.

### Crashlog

I got an exeption : "Gatt SetCharacteristicNotification FAILED."

i check in source code and found that :

> / In order to subscribe to notifications on a given characteristic, you must first set the Notifications Enabled bit
>             // in its Client Characteristic Configuration Descriptor. See https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorsHomePage.aspx and
>             // https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorViewer.aspx?u=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
>             // for details.

i dont know why 
```
if (!_gatt.SetCharacteristicNotification(_nativeCharacteristic, true))
                throw new CharacteristicReadException("Gatt SetCharacteristicNotification FAILED.");
```
return false :/

## Configuration

**Version of the Plugin:** e.g. 1.2.1 

**Platform:** e.g. Android 4.4.2

**Device:** e.g. wiko rainbow 4G

xabre, 2017-03-23:
Can you try a third party ble explorer app (like  [BLE Scanner](https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner) [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp)) to try to connect discover your services/characteristics write the data and then start notifications. Please confirm that the notifications work when using these apps on your phone.

joriano, 2017-03-23:
Hi,

I confirm that with nrf connect i can write my characteristic.
I only can update with notification enable, in nrf connect, for my RX characteristic.

I think I just need to enable notification with BLE plugin but I don’t know how to do that.

Regards

Jean Oriano
R&D Prastel France
Email : joriano@prastelfrance.fr<mailto:joriano@prastelfrance.fr>
Tel : 04.42.98.06.06

[cid:image001.jpg@01D04A26.C869BD50]

PRASTEL France
Z.I. Athélia II
225, Impasse du Serpolet
13704 La Ciotat Cedex - France
Tél : +334.4298.0606
Fax : +334.4204.5351

[qrcode_Pierre Jaumes]

www.prastel.com<http://www.prastel.com/>
Découvrez notre nouveau catalogue !

[PRASTEL catalogue n.10_FR_COUV-mini]<http://www.prastel.com/upload/actualites/nouveau-catalogue-7.10_fr_ld.pdf>
Ou clickez sur ce lien : catalogue PRASTEL n°10<http://www.prastel.com/upload/actualites/nouveau-catalogue-7.10_fr_ld.pdf>

joriano, 2017-03-29:
hi,
I come with more information.
I tried to notify my characteristic with the BLE explorer application. "start update" button is ok and launch notification. but nothing is receive.
My NRF52 send data over NUS nordic service.

I tried with nrfConnect, and i receive data to my notified characteristic.

When i implement my proper application, i'm connected with my NRF52, i discover UART service and characteristic and i can write one.
I can startUpdate but nothing appear, no error, no exeptions, no received data.
My NRF52 send data over NUS nordic service.

see this piece of source code.

update function :
`public async void StartUpdateRXCharacteristic(byte[] dataRX)
        {
            try
            {
                _UartRXcharac.ValueUpdated -= CharacteristicOnValueUpdated;
                _UartRXcharac.ValueUpdated += CharacteristicOnValueUpdated;
                await _UartRXcharac.StartUpdatesAsync();
                Debug.WriteLine("######################## START UPDATE OK ##############");
            }
            catch (Exception ex)
            {
                Debug.WriteLine("######################## ERROR ##############");
                Debug.WriteLine("########## {0} ", ex.Message);
            }
        }`

write function :
`        public async void WriteToTXCharacteristic(byte [] dataTX)
        {
            Debug.WriteLine("######################## START WRITE CHARACTERISTIC ##############");

            if (_UartTXcharac.CanWrite)
                await _UartTXcharac.WriteAsync(dataTX);
        }`

function where i try to communicate with NRF52 (on my nrf52 i only get data over ble and when i occured, i send an answer.
`        public async void Adapter_DeviceConnected(object sender, DeviceEventArgs e)
        {
            Debug.WriteLine($"############ Connected {e.Device.Name}");

            StopScanDevice();
            _deviceConnected = e.Device;

            //récupère le service UART nordic
            _UartService = await GetUartService(e.Device);
            //récupère les canaux RX et TX du service UART
            _UartTXcharac = await GetTXcharacteristic();
            _UartRXcharac = await GetRXcharacteristic();

            byte[] data = new byte[5];
            byte[] data1 = new byte[5];
            data1[0] = 0x40;
            data1[1] = 0x45;
            data1[2] = 0x50;

            WriteToTXCharacteristic(data1);
            StartUpdateRXCharacteristic(data);
        }`

Regards

tomridl, 2017-03-31:
@joriano Maybe double check that the characteristic supports updates?

```
if (_UartRXcharac.CanUpdate)
{
    await _UartRXcharac.StartUpdatesAsync();
}
else
{
    Debug.WriteLine("Updates not supported");
}
```

joriano, 2017-04-10:
Hi,

It is my new update function.
I havnt notification again :/

```
public async void StartUpdateRXCharacteristic(byte[] dataRX)
        {
            try
            {
                _UartRXcharac.ValueUpdated -= CharacteristicOnValueUpdated;
                _UartRXcharac.ValueUpdated += CharacteristicOnValueUpdated;
                
                if (_UartRXcharac.CanUpdate)
                {
                    await _UartRXcharac.StartUpdatesAsync();
                    Debug.WriteLine("######################## START UPDATE OK ##############");
                }
                else
                {
                    Debug.WriteLine("Updates not supported");
                }
            }
            catch (Exception ex)
            {
                Debug.WriteLine("######################## ERROR ##############");
                Debug.WriteLine("########## {0} ", ex.Message);
            }
        }
```

```
        private void CharacteristicOnValueUpdated(object sender, CharacteristicUpdatedEventArgs characteristicUpdatedEventArgs)
        {
            dataRX = characteristicUpdatedEventArgs.Characteristic.Value;
            Debug.WriteLine("New value: ");
        }
```

Tijgerd, 2017-09-26:
Same is happening for me.
If I look with a bluetooth terminal app I do receive updates from my BLE device.
The Characteristic.ValueUpdated handle never triggers :(

tcatalano, 2018-08-02:
I know this is an old thread but I am having same problem with Characteristic.ValueUpdated not firing. Have verified with nrfConnect that notification is definitely being sent from device. Have tried 1.3 and 2.0 packages.

Was there a resolution? Thanks for any info.

vimalaven, 2020-06-08:
Same, it's not working for me too. The event never gets triggered. Any idea?