dotnet-bluetooth-le/dotnet-bluetooth-le #177: ValueUpdated event not fired on Android for a notify characteristic

shayo, 2017-03-25:
I am running an app on Xamarin Forms and using your plugin to communicate with a Nordic NRF51 module. Overall, everything works fine. I can send commands and get notifications on both iOS and Android.
The problem started when I tried to run code that runs device firmware update (FOTA).
This procedure requires to send some commands to a control characteristic, and then wait for a reply notification over another. 
The shared code I'm using works without a problem on iOS devices, but fails on Android devices because I never get that notification.
This is peculiar because for all other characteristics, I do get notifications back from the Android device.

I'm using version 1.2.1.

I know this isn't much to work on (specially because it only doesn't work on this very specific instance), but what could possibly cause this?
Is there some hint / lead I can follow in your code to assess why notifications are not received for this characteristic? (again - everything works fine in iOS, and everything works fine for other characteristics in Android).

closed 2017-03-25 by shayo

cipriancoj, 2017-04-18:
I had the same problem with an old version of these apis but with the last one I solved it!

xabre, 2017-04-24:
Which version are you using now? 1.2.3 or 1.3.0-alpha ?

cipriancoj, 2017-04-24:
Now I'm using the 1.2.3