dotnet-bluetooth-le/dotnet-bluetooth-le #178: Notification not fired on Android device

shayo, 2017-03-25:
I am running an app on Xamarin Forms and using your plugin to communicate with a Nordic NRF51 module. Overall, everything works fine. I can send commands and get notifications on both iOS and Android.
The problem started when I tried to run code that updated the device firmware (DFU FOTA).
This procedure requires to send some commands to a control characteristic, and then wait for a reply notification that those were ack' by the device.
The shared code I'm using **works without a problem on iOS** devices, but fails on Android devices (I never get that notification).
This is peculiar because for **all other characteristics, I do get notifications back from the Android device.**

I know this isn't much to work on (specially because it only happens on this very specific case), but what could possibly cause this?
Is there some hint / lead I can follow in your code to assess why notifications are not received for this characteristic? (again - everything works fine in iOS, and everything works fine for other characteristics in Android).

## Steps to reproduce

1. Get yourself a nordic NRF51DK module that runs a bootloader from SDK 10.
2. Put device in DFU mode 
3. Start DFU update.

## Expected behavior
dfuCharacteristicControlPoint should get notified with a response.

## Actual behavior
dfuCharacteristicControlPoint notification never gets fired on Android devices.
On iOS it works fine.

### Crashlog

No errors on application output.

This is mostly my debugging information:

DFUTarg: Start DFU of device '{0}'
Read data... 
08-00: Data read ({0})
DFU service revision: 8
[BluetoothGatt] setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
[Mono] DllImport searching in: '__Internal' ('(null)').
[Mono] Searching for 'java_interop_jnienv_new_byte_array'.
[Mono] Probing 'java_interop_jnienv_new_byte_array'.
[Mono] Found as 'java_interop_jnienv_new_byte_array'.
[Mono] DllImport searching in: '__Internal' ('(null)').
[Mono] Searching for 'java_interop_jnienv_set_byte_array_region'.
[Mono] Probing 'java_interop_jnienv_set_byte_array_region'.
[Mono] Found as 'java_interop_jnienv_set_byte_array_region'.
Thread started:  #13
Resolved pending breakpoint at 'DfuController.cs:351,49' to void NordicDFU.DfuController.
Resolved pending breakpoint at 'DfuController.cs:351,49' to void NordicDFU.DfuController.<StartDFU>c__async4.MoveNext () [0x00363].
Resolved pending breakpoint at 'DfuController.cs:357,1' to void NordicDFU.DfuController.<StartDFU>c__async4.MoveNext () [0x00387].
Resolved pending breakpoint at 'DfuController.cs:364,1' to void NordicDFU.DfuController.<StartDFU>c__async4.MoveNext () [0x003ab].
Resolved pending breakpoint at 'DfuController.cs:298,1' to void NordicDFU.DfuController.<StartDFU>c__async4.<StartDFU>c__AnonStorey10.<>m__0 (object sender, Plugin.BLE.Abstractions.EventArgs.CharacteristicUpdatedEventArgs e) [0x00001].
Start DFU...
01-04: Write data... ({0})
Data written
00-00-00-00-00-00-00-00-E0-FC-00-00: Write data... ({0})
Data written

**[ At this point, iOS sends a notification back on this characteristic]
On Android, it never arrives.**

## Configuration

**Version of the Plugin:** 1.2.1
**Platform:**  Android 5.1
**Device:** Moto G

xabre, 2017-03-31:
It's difficult to say.... How many characteristic subscriptions do yo have? The number is limited...

xabre, 2019-08-11:
closed due to inactivity

closed 2019-08-11 by xabre