dotnet-bluetooth-le/dotnet-bluetooth-le #179: Example app doesnt work

Hussamalsh, 2017-03-27:
Hi

When i run the example app on my android device  (Samsung galaxy s7) i got the following error:

Unhandled Exception:
System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.

on line 33 in MainActivity.cs    ( Mvx.Resolve<IMvxAppStart>().Start();)

## Configuration

**Platform:** / Android 

**Device:** / Samsung galaxy s7

Best regards Hussam

xabre, 2017-03-31:
@Hussamalsh I've just checked out and deployed the current state of the sample app form the master branch. Seems to be running fine. 

Please check your Android SDK Manager, maybe something is missing there or out of date. Do a clean build after you install the needed stuff. Moreover, please try to include a more detailed stack trace.

xabre, 2017-04-22:
Closing issue

closed 2017-04-22 by xabre