dotnet-bluetooth-le/dotnet-bluetooth-le #180: Android Exception: GATT Write Characteristic Failed

tomridl, 2017-03-29:
Intermittent issue, after BT connection is complete, I attempt to write to a characteristic on the remote device, and in some instances (approx 50% of the time), an exception is thrown.

## Steps to reproduce

1. Scan for available BT devices
2. Connect to device
3. GetServiceAsync for a specific characteristic
4. Assign event handler for ValueUpdated event for that characteristic
5. If characteristic CanUpdate is true, StartUpdatesAsync
6. Attempt to write data to that characteristic using WriteAsync

## Expected behavior
Data is successfully written to the characteristic. In this case, this is the method of transparent data communication between the app and the BT device.

## Actual behavior
An exception occurs in approximately 50% of connections

### Crashlog

[mono] Unhandled Exception:
[mono] Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED.
[mono]   at Plugin.BLE.Android.Characteristic.InternalWrite (System.Byte[] data) [0x0004a] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:99 
[mono]   at Plugin.BLE.Android.Characteristic+<>c__DisplayClass19_0.<WriteNativeAsync>b__0 () [0x00000] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:75 
[mono]   at Plugin.BLE.Abstractions.Utils.TaskBuilder+<FromEvent>d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext () [0x000b3] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\Utils\TaskBuilder.cs:50 
[mono] --- End of stack trace from previous location where exception was thrown ---
[mono]   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
[mono]   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00047] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187 
[mono]   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156 
[mono]   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128 
[mono]   at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:357 
[mono]   at Plugin.BLE.Android.Characteristic+<WriteNativeAsync>d__19.MoveNext () [0x00097] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:74 
[mono] --- End of stack trace from previous location where exception was thrown ---
[mono]   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
[mono]   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00047] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187 
[mono]   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156 
[mono]   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128 
[mono]   at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:357 
[mono]   at Plugin.BLE.Abstractions.CharacteristicBase+<WriteAsync>d__31.MoveNext () [0x0006c] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\CharacteristicBase.cs:91 
[mono] --- End of stack trace from previous location where exception was thrown ---
[mono]   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionse
[mono-rt] [ERROR] FATAL UNHANDLED EXCEPTION: Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED.
[mono-rt]   at Plugin.BLE.Android.Characteristic.InternalWrite (System.Byte[] data) [0x0004a] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:99 
[mono-rt]   at Plugin.BLE.Android.Characteristic+<>c__DisplayClass19_0.<WriteNativeAsync>b__0 () [0x00000] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:75 
[mono-rt]   at Plugin.BLE.Abstractions.Utils.TaskBuilder+<FromEvent>d__1`3[TReturn,TEventHandler,TRejectHandler].MoveNext () [0x000b3] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\Utils\TaskBuilder.cs:50 
[mono-rt] --- End of stack trace from previous location where exception was thrown ---
[mono-rt]   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
[mono-rt]   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00047] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187 
[mono-rt]   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156 
[mono-rt]   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128 
[mono-rt]   at System.Runtime.CompilerServices.TaskAwaiter`1[TResult].GetResult () [0x00000] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:357 
[mono-rt]   at Plugin.BLE.Android.Characteristic+<WriteNativeAsync>d__19.MoveNext () [0x00097] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Android\Characteristic.cs:74 
[mono-rt] --- End of stack trace from previous location where exception was thrown ---
[mono-rt]   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionservices/exceptionservicescommon.cs:143 
[mono-rt]   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x00047] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:187 
[mono-rt]   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x0002e] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:156 
[mono-rt]   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x0000b] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/compilerservices/TaskAwaiter.cs:128 
[... cut ...]
[mono-rt]   at Plugin.BLE.Abstractions.CharacteristicBase+<WriteAsync>d__31.MoveNext () [0x0006c] in C:\Users\Adrian\Documents\GitHub\xamarin-bluetooth-le\Source\Plugin.BLE.Abstractions\CharacteristicBase.cs:91 
[mono-rt] --- End of stack trace from previous location where exception was thrown ---
[mono-rt]   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in /Users/builder/data/lanes/4009/3a62f1ea/source/mono/mcs/class/referencesource/mscorlib/system/runtime/exceptionserv

## Configuration

**Version of the Plugin:** 1.2.1

**Platform:** Android 7.1.1

**Device:**  Google Pixel

shayo, 2017-03-30:
I am experiencing something similar. Also under Android. Trace follows:

caused.AggregateException(s)
Plugin.BLE.Android.Characteristic.InternalWrite(byte[] data)<06ecde4ffed24ee08d2a79e5007a1ee1>:0
Plugin.BLE.Android.Characteristic.<>c__DisplayClass19_0.<WriteNativeAsync>b__0()<06ecde4ffed24ee08d2a79e5007a1ee1>:0
Plugin.BLE.Abstractions.Utils.TaskBuilder.<FromEvent>d__1<TReturn, TEventHandler, TRejectHandler>.MoveNext()<aa2c9c7c6f724aaba139a63bffca5561>:0
Plugin.BLE.Android.Characteristic.<WriteNativeAsync>d__19.MoveNext()<06ecde4ffed24ee08d2a79e5007a1ee1>:0
Plugin.BLE.Abstractions.CharacteristicBase.<WriteAsync>d__31.MoveNext()<aa2c9c7c6f724aaba139a63bffca5561>:0
--- End of inner exception stack trace ---
---> (Inner Exception #0) Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: Gatt write characteristic FAILED.
Plugin.BLE.Android.Characteristic.InternalWrite(byte[] data)<06ecde4ffed24ee08d2a79e5007a1ee1>:0
Plugin.BLE.Android.Characteristic.<>c__DisplayClass19_0.<WriteNativeAsync>b__0()<06ecde4ffed24ee08d2a79e5007a1ee1>:0
Plugin.BLE.Abstractions.Utils.TaskBuilder.<FromEvent>d__1<TReturn, TEventHandler, TRejectHandler>.MoveNext()<aa2c9c7c6f724aaba139a63bffca5561>:0
Plugin.BLE.Android.Characteristic.<WriteNativeAsync>d__19.MoveNext()<06ecde4ffed24ee08d2a79e5007a1ee1>:0
Plugin.BLE.Abstractions.CharacteristicBase.<WriteAsync>d__31.MoveNext()<aa2c9c7c6f724aaba139a63bffca5561>:0

This is crashing the app even though it is in a try catch block.

tomridl, 2017-03-30:
Additional info: This seems to occur less often when the WriteAsync is within a Device.BeginInvokeOnMainThread, so perhaps this is a threading issue?

xabre, 2017-03-30:
There is a limitation in android which requires all ble writes to be called from the main thread. I.e. whenever you call await something async call it from the main thread

xabre, 2017-03-30:
Moreover, make sure to await your calls, don't use await in methods that return void. Ble commands should not overlap, write might fail because the adapter is busy with some other command.

ClausElmann, 2017-03-31:
It would be nice if you are able to catch it in a try catch block, not just crash the app.

tomridl, 2017-03-31:
So I've updated my code based on comments from @xabre, to the following:
```

        Device.BeginInvokeOnMainThread(async() => { await SendData(data);});

        private static async Task<bool> SendData(byte[] data)
        {
            bool result = false;

            try
            {
                result = await TransparentDataCharacteristic.WriteAsync(data);
            }
            catch (Exception ex)
            {
                Debug.WriteLine("Write fail: " + ex.Message);
            }

            return result;
        }

```
Exceptions are caught by the try catch and handled.

One problem I found was that I was sending data in the OnAppearing event of a tabbed page, but there is a Xamarin bug where the OnAppearing event is called twice when tabbed pages are loaded, so it was double calling the send too quickly, this seemed to generate the error. If I try and call send to quickly, even with the awaits, it still generates a GATT error.

ClausElmann, 2017-03-31:
@tomridl 

Can I se how you declare TransparentDataCharacteristic as SendData is static

tomridl, 2017-03-31:
@ClausElmann 

```
        public static ICharacteristic TransparentDataCharacteristic
        {
            get; set;
        }
```

After a device is connected, get the characteristic either by discovering all characteristics, or else get a specific characteristic as so:

`TransparentData.TransparentDataCharacteristic = await transparent.GetCharacteristicAsync(Guid.Parse("49535343-1E4D-4BD9-BA61-23C647249616"));
`

xabre, 2017-03-31:
If any method which calls 
`Device.BeginInvokeOnMainThread(async() => { await SendData(data);});` gets called again (like on appear) you will get the threading issue and subsequently the error. You should make sure that your code always completes a ble task before invokeing the next.

I was thinking of introducing a helper class, a serial queue which could be used to guarantee that async tasks called by using it are executed  serially. I could also implement the main thread logic in there. This might help developers avoid this limitation.

ClausElmann, 2017-03-31:
Sounds great, if you already have some sample code, I would like a copy👍

aliraza-cinnova, 2018-08-07:
I am having the same problem and applied all of the above solutions. But still most of the times, issue is there. 5% chances are there that during debugging, it shows correct data loaded. HELP

xabre, 2019-08-11:
#276 Similar

closed 2019-08-11 by xabre

PippoApps, 2022-02-25:
It seems today still Android cannot write to writeWithoutResponse characteristics... do you confirm?

ManikandanK-AMI, 2022-09-08:
Hi

I'm still facing the same issue

https://github.com/xabre/xamarin-bluetooth-le/issues/621#issuecomment-1239658780

Please help me and provide the solution