dotnet-bluetooth-le/dotnet-bluetooth-le #182: Autoconnect on Android

xabre, 2017-03-31:
Investigate and document the behaviour of Android the auto-connect flag.

AnoopChatra, 2017-04-03:
Hi ,

    I am not at all able to re-connect  to the BLE device using xamarin-bluetooth-le plugin, though "auto-connect" flag is set to true. I have tested my BLE devices with the "native android mobile apps" and "mono android mobile apps" the  auto-connect / auto-reconnect works perfectly fine in either of the cases.

In xamarin-bluetooth-le plugin I am using the following method to connect to the BLE device and I have successfully connected:
CrossBluetoothLE.Current.Adapter.ConnectToDeviceAsync(deviceToBeConnected,new ConnectParameters(true,false)) ;
Here the first param of ConnectParameters is "AutoConnect" flag , which needs to be set to true for auto re-connections.

Inside ConnectToDeviceAsync() method,  ConnectGatt() will be called: 
((BluetoothDevice)device.NativeDevice).ConnectGatt(Application.Context, connectParameters.AutoConnect, _gattCallback) 
Here _gattCallback is the implementation of BluetoothGattCallback. It contains overridden method OnConnectionStateChange() which gets the BLE connection change status.

Inside  OnConnectionStateChange() on ProfileState.Disconnected " gatt.Close();" is getting called.When close() is called I believe the "gatt" object will be deallocated and it wont be available anymore .So there is no chance of re-connection.

Is it really required to call  " gatt.Close();" on  ProfileState.Disconnected change ? Is there any reason behind it? 
I guess auto-connection will work if  we do not close the Gatt object. Could you please help me out to solve this issue ?? 

Regards,
Anoop Chatra

AnoopChatra, 2017-05-04:
Hey @xabre ,

      Any updates about auto connect??

AnoopChatra, 2017-05-16:
Hi @xabre

   I have pulled the latest code which  contains the fix for "auto reconnection" in android. I have built the dlls locally and I am using it in my application.
   I can successfully connect and reconnect to the BLE device but the I am getting "CharacteristicReadException" while reading the characteristics . Please refer the below log for more info:
   
   05-16 15:47:48.460 I/mono-stdout(14184):  BleAdapter : read characteristic : xyzxyzxyz-xyzx-xyzx-xyzxyzxyz
BleAdapter : CharacteristicReadException: xyzxyzxyz-xyzx-xyzx-xyzxyzxyz   at Plugin.BLE.Android.Characteristic.ReadInternal () 
[0x0001a] in D:\PluginBLE\AutoConnect\xamarin-bluetooth-le-ac4398b855dac6d05b39d9d24cb04263f3429dc9\Source\Plugin.BLE.Android\Characteristic.cs:72

Regards,
Anoop Chatra

xabre, 2017-05-18:
I still need to determine the proper way of disposing the IDevice memebrs in case of autoconnect/connectionloss/reconnect ... I will get some time to work on this hopefully during the next weeks. :)

AnoopChatra, 2017-05-22:
@xabre ,
May I know next release dates??  Is "AutoConnect" is going to be incorporated in the next release?

xabre, 2017-07-23:
I didn't get the time lately to work on the plugin because of work/personal issues. Any PR are wellcome and will be merged a.s.a.p.

PR #519 (2022-03-19): Fix Android autoConnect flag
Resolves #182 

Basically only closes the gatt instance when autoConnect was set to false when connecting to that device
closed 2022-03-19 by janusw