dotnet-bluetooth-le/dotnet-bluetooth-le #183: UpdateRssiAsync() skip code

SeanStayn, 2017-04-03:
Hello,

my case:
I update the RSSI value from my device with the method device.UpdateRssiAsync(). With an if-condition I fire an event, if the method return true, like the snippet below.

Code snippet:
var rssiUpdateResult = await device.UpdateRssiAsync();

if (rssiUpdateResult ){
                        this.RssiUpdateEvent?.Invoke(this, new RssiUpdateEventArgs(device));
}
Debug.WriteLine("RSSI - UpdateMethodCallFinished : " + rssiUpdateResult;

After the if-condition, I send a message to the console.

The problem:
Every line of code, after the method don't execute. I set breakpoints and the programm stops at the line of the UpdateRssiAsync method call, but it doesn't stop at the if-condition or the WriteLine method. Additionally I can't find any output from the WriteLine method.

Question:
What do I wrong? I can make a workaround, if I fire my event before I call the UpdateRssiAsync method, but I think it's not the right way.

tomridl, 2017-04-11:
Same issue on Samsung J5 (Android 5.1.1), plugin version 1.2.2. 

When I tested on a Pixel (Android 7.1.2), the code completed, but the RSSI value was never updated (remained the same as when the unit first connected, so moving the phone away from the unit did not cause a change in RSSI value). On iOS, the RSSI value dynamically updates when moving the phone closer or nearer the device.

BradleyAlcock, 2017-04-13:
xamarin-bluetooth-le/Source/Plugin.BLE.Android/Device.cs

Line 206 if (args.Device.Id == Id) return;
this will always return based on my understanding of the code.

suggest enclosing the the other if/else within like this:

if (args.Device.Id == Id)
				  {

					  if (args.Error == null)
					  {
						  Trace.Message("Read RSSI for {0} {1}: {2}", Id, Name, args.Rssi);
						  Rssi = args.Rssi;
						  complete(true);
					  }
					  else
					  {
						  Trace.Message($"Failed to read RSSI for device {Id}-{Name}. {args.Error.Message}");
						  complete(false);
					  }
				  }

tomridl, 2017-04-13:
Or:

` if (args.Device.Id != Id) return;`

BradleyAlcock, 2017-04-13:
yep, that does work better, the return is used to stop returning the same rssi value again.

xabre, 2017-04-18:
yes, you are right, problem got fixed in 1.3.0-alpha. 
I will also include it in a 1.2.3 release until we confirm that 1.3.0 is stable.

xabre, 2017-04-18:
Fixed with v1.2.3 :)

closed 2017-04-18 by xabre

helderdb, 2020-10-20:
It's happening in 2.1.1 as well..

daniel-streetec, 2021-09-20:
Hi, I can confirm that using 2.1.2 and Samsung Galaxy S10 the function will never return and any code execution is blocked.

`if (this.BLEDevice.State == Plugin.BLE.Abstractions.DeviceState.Connected)
            {
                await this.BLEDevice.UpdateRssiAsync();
                return this.BLEDevice?.Rssi ?? -1;
            } else
            {
                return -1;
            }`