dotnet-bluetooth-le/dotnet-bluetooth-le #187: Characteristic.WriteAsync never returns on Android

mathiasnielsen, 2017-04-06:
Hi guys!

Saw issue #39, and I am experiencing the same, but for android (but only on my Huawei mate 9 pro, not other android devices).

## Steps to reproduce
1. Connect to device
2. Discover services
3. Discover characteristic
4. Write to characteristic async

The code is as follows:

            SynchronizationContext.Current.Post(
                async (param) =>
            {
                var result = await _characteristic.WriteAsync(mybytes)
                completionSource.TrySetResult(result);
            }, null);

... if I, add a Task.Delay(2000) before this, it seems to work.

## Expected behavior
The "WriteAsync" method returns at some point.

## Actual behavior
It never returns.

## Configuration

**Version of the Plugin:** ... from approx Feb 2017

**Platform:** Android 7.0

**Device:** Huawei Mate 9 Pro

xabre, 2017-04-07:
Well, given the above code, in case WriteAsync fails with an exception your tasc completions ource will never return. You can try something like this:

  SynchronizationContext.Current.Post(
            async (param) =>
        {
           try {
            var result = await _characteristic.WriteAsync(mybytes)
            completionSource.TrySetResult(result);
         }catch(Exception ex){
            completionSource.TrySetException(ex);
        }
        }, null);

mathiasnielsen, 2017-04-11:
Good advice - but it does not seem to make a difference... It works with a 2000ms delay, or if I have a breakpoint on the "WriteAsync" method, for a short period of time, before it gets executed. Any other ideas?

xabre, 2017-04-14:
Can you give us more information on your use case? Maybe a minimum working code sample so that we can reproduce the problem?

mathiasnielsen, 2017-04-19:
sure thing - sadly I'm pretty busy at the moment, but will provide you with one within a week.

tototrix, 2018-06-22:
Hi,
I have the same problem. Have you resoled it ?

juCortez, 2019-10-23:
same problem read/write characteristic never returns...