dotnet-bluetooth-le/dotnet-bluetooth-le #188: IsOn and IsAvailable always return false and true on iOS regardless of bluetooth state

Yelnik, 2017-04-07:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Check CrossBluetoothLE.Current.IsOn and CrossBluetoothLE.Current.IsAvailable on iOS and both values will return false and true respectively no matter if the bluetooth is on or off on the device

## Expected behavior
I expect IsOn to return true if the bluetooth adapter is turned on

## Actual behavior
IsOn always returns false

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
1.2.2

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
10.2.1

**Device:** e.g. HTC Sensation /i Phone 7 ...
iPhone 5C

smstuebe, 2017-04-10:
- Have you confirmed this behavior with the sample app?
- If you subscribe to the `StateChanged` event, does it fire, when you turn on / off BLE?

Yelnik, 2017-04-10:
The sample app I don't know.

I wasn't worried about state change, just the state when the app is being opened, which unfortunately is always false

smstuebe, 2017-04-10:
The Sample App is contained in this repository. See: https://github.com/xabre/xamarin-bluetooth-le/tree/master/Source/BLE.Client 

Did you somehow click "Don't allow" when the permission dialog popped up?
Can you try do run the sample app?

tomridl, 2017-04-11:
What I did was put a check in the OnAppearing method of the first page that gets loaded to see if BT is turned on. The first time that the page loads, CrossBluetoothLE.Current.IsOn returns false even if BT is turned on. However, on any subsequent pages, or even returning to the initial page, it returns the correct value (true if BT turned on, false if BT turned off). If, however, the check for IsOn is on the second page of an application, the correct value is returns immediately. 

I then put a 1 second timer in the initial OnAppearing method, and IsOn then returned the correct value on the first page. So it seems to be a timing issue, if you read IsOn as soon as the application starts, it returns the incorrect value.

erdmenchen, 2017-04-18:
Too bad, I am experiencing the same issue with "IsOn". 

I have an application that starts to scan for nearby BLE advertisements in the "OnAppearing" method of the main view.

Delaying the call with a timer is a hotfix solution to the timing issue. 
However this cannot be the final solution when using this package in production...

xabre, 2017-04-25:
Are you doing a check before scanning? For example:
```
if(IsOn){
   scan()
}
```

You might try to do something like:

```
if(IsOn){
   scan()
}
else{
 _ble.StateChanged += (o, a) =>{
     if(a.State == ON)
        scan();
}
}
```

Moreover on iOS it's enough to call scan even if state is off, because the actual scanning will only start once the state changes to ON.

Yelnik, 2017-04-26:
The scanning is fine, it was just determining on launch if Bluetooth is turned on

Duality4Y, 2017-06-23:
Hi, I am experiencing the same thing, I check ```IsOn``` before I do anything, that works fine on android but not on ios.

CarlesBaldovi, 2017-12-01:
Is there anything new about this?

AlexusBlack, 2018-04-13:
Okay, so I checked the Plugin.BLE source code and played with it a bit and found an inconsistency between Android and iOS state detection that Plugin.BLE does not fully compensate.

While on Android we can get state immediately, on iOS initial state is unknown and delivered later via state update event. This is alike CBCentralManager which also delivers states via a delegate. 

I wrote a simple workaround which is a solution for issues #188 and #112. I tested it on both iOS and Android, using iOS 11 IP7 and Android 7 Samsung Galaxy S7.

```csharp
/// <summary>
/// Checks the bluetooth adapter if it is on or off.
/// </summary>
private async Task CheckBluetooth() {
    // iniial state check
    var state = await GetBluetoothState(Plugin.BLE.CrossBluetoothLE.Current);
    if(state == BluetoothState.Off)
    {
        ShowDisabledBluetoothAlert();
    }
    
    // tracking state in case if user disable his bluetooth
    Plugin.BLE.CrossBluetoothLE.Current.StateChanged += (o, e) =>
    {
        if(e.NewState == BluetoothState.Off)
        {
            ShowDisabledBluetoothAlert();
        }
    };
}

/// <summary>
/// Get current bluetooth state
/// </summary>
/// <param name="ble">Crossplatform bluetooth implementation object</param>
/// <returns>BluetoothState</returns>
private Task<BluetoothState> GetBluetoothState(IBluetoothLE ble)
{
    var tcs = new TaskCompletionSource<BluetoothState>();

    if (ble.State != BluetoothState.Unknown)
    {
        // If we can detect state out of box just returning in
        tcs.SetResult(ble.State);
    } else
    {
        // Otherwise let's setup dynamic event handler and wait for first state update
        EventHandler<Plugin.BLE.Abstractions.EventArgs.BluetoothStateChangedArgs> handler = null;
        handler = (o, e) =>
        {
            Plugin.BLE.CrossBluetoothLE.Current.StateChanged -= handler;
            // and return it as our state
            // we can have an 'Unknown' check here, but in normal situation it should never occur
            tcs.SetResult(e.NewState);
        };
        Plugin.BLE.CrossBluetoothLE.Current.StateChanged += handler;
    }

    return tcs.Task;
}

/// <summary>
/// Show alert to user that his bluetooth is disabled
/// </summary>
private void ShowDisabledBluetoothAlert()
{
    App.ShowAlert("Bluetooth disabled", "Bluetooth required to work with BLE devices.");
}
```

volcano619, 2023-09-20:
This seems to be still present

alifindik10, 2025-03-02:
This error is still present but workaround works.